serialport/node-serialport #1579: [Bug Report] Sending string with ° or ³, randomly get ��

KingMario, 2018-06-22:
- __SerialPort Version__: 6.2.0
- __NodeJS Version__: 8.11.3
- __Operating System__ and __Hardware Platform__: macOS High Sierra 10.13.4

- Have you checked the right __version__ of the __api docs?__: yes.
- Are you having trouble installing and you checked the __Installation Special Cases__ docs?: no.
- Are you using __Electron__ and have you checked the _Electron Docs_?: no.

### Summary of Problem
Send string with ° or ³, randomly get ��

### Steps and Code to Reproduce the Issue
Two serial ports connected using a loop cable. One is opened for reading, and the other is opened for writing. While writing a string with ° or ³, from the reading port, randomly get ��.

writing code:
```js
port.write(`<?xml version="1.0" encoding="UTF-8" standalone="yes" ?><MT><temp_set>20.00 °C</temp_set><d>0.99940 g/cm³</d></MT>
```

reading code:
```js
const parser = new RegexParser({
  regex: /<\/MT>/
})
port.pipe(parser)
parser.on('data', function (data) {
  const rawData = `${data}</MT>`
  console.log(rawData)
})    
```

KingMario, 2018-06-22:
Looks like it's a bug of RegexParser.

```
const parser = new Readline({
  delimiter: '</MT>'
})
```

Changing the parser to Readline fixes the problem.

reconbot, 2018-08-29:
You might want to try specifying an encoding the regex parser? If you can make a failing test case for it I'd be happy to fix the bug.

closed 2018-08-29 by reconbot