serialport/node-serialport #1584: Problems releasing v6.2.1

reconbot, 2018-06-28:
- __SerialPort Version__: v6.2.1

I'm getting hit by this bug on master https://github.com/mafintosh/prebuildify/issues/10 and then in my v6.3.2 tag I've changed to building on node 10 and I'm getting hit by this bug https://github.com/prebuild/prebuild/issues/221

I think I need to go back to node 8 for builds and then set `build_v8_with_gn=false` in the configure script run by prebuild but I have no idea how to do that.

The issue with building for node10 when running node 8 is easily reproducible locally by running `npm run prebuild`. I have to come back to this as I've spent more on this issue than I should.

I haven't npm released v6.2.1 because I keep changing some of the code behind it for build purposes. I would *love* to get it out.

ralphtheninja, 2018-06-28:
@reconbot Let me know how I can help.

reconbot, 2018-06-29:
Can you figure out how to get node 8 to prebuild node 10?

ralphtheninja, 2018-06-29:
Are you using `prebuildify` or `prebuild`?

reconbot, 2018-06-29:
Prebuild (see our package.json) but it's a similar issue.

ralphtheninja, 2018-06-29:
> Prebuild (see our package.json) but it's a similar issue.

Aye, noticed prebuild in package.json. Was just a bit confused when you mentioned prebuildify.

Anyway, I personally don't have these problems when building locally by doing e.g. `npm run prebuild` on latest master. Meaning I'm running node v10.5.0 and `npm run rebuild` builds all binaries. I'm on linux though and installing node via nvm (not sure that matters, but just explaining my environment).

ralphtheninja, 2018-06-29:
@reconbot Another alternative is to use `prebuild-ci` and move away from the `BINARY_BUILDER` thing. We have used it in `leveldown` for quite some time now and it works really well. Let me know if you need me to walk you through it.

reconbot, 2018-06-29:
I'd love to know more about it, I'll be able to dig in later on but any
direction you can give would be great.

reconbot, 2018-06-30:
I dislike that `prebuild --all` is broken when it builds for node 10 from node 8, but I guess this works too. 🤷‍♀️ 

I've updated the configs to support `prebuild-ci`'s new setup and successfully released `v6.2.1` 🎉

Thanks for the tip!

closed 2018-06-30 by reconbot

ralphtheninja, 2018-06-30:
> I dislike that prebuild --all is broken when it builds for node 10 from node 8, but I guess this works too. woman_shrugging

I also dislike this, but difficult to do something _easy_ about it imo.

> Looks like prebuild-ci will upload whatever version is running when the command is invoked (eg, that version of the os and node)?

That's correct!