serialport/node-serialport #1588: Missing prebuild binaries in 6.2.1

monkbroc, 2018-07-01:
- __SerialPort Version__: 6.2.1
- __NodeJS Version__: 5.4.1
- __Operating System__ and __Hardware Platform__: Windows 10 x64

### Summary of Problem

In version 6.2.0, `serialport` included `prebuild` binaries for Node 4.X, 5.X, 6.X, 7.X, 8.X and 9.X. [See release on GitHub](https://github.com/node-serialport/node-serialport/releases/tag/v6.2.0).

In version 6.2.1, `serialport` includes `prebuild` binaries for Node 6.X, 8.X and 10.X. [See release on GitHub](https://github.com/node-serialport/node-serialport/releases/tag/v6.2.1)

It is surprising to see a change that breaks the installation of the package for end users with older Node runtimes in a patch release.

I understand that Node 4.X, 5.X and 7.X are at end of life, but I'm not sure if the intent of removing `prebuild` binaries for those versions means that `serialport` does not support those Node versions anymore.

If Node 4.X, 5.X and 7.X are not supported, it would be nice to update the [Platform Support section of the README.md](https://github.com/node-serialport/node-serialport/blob/fb565fd345224be34080be598329c739dfaa7925/README.md#platform-support) to say so. I would have liked to see dropping support for Node versions as a breaking semver change, not a patch change.

reconbot, 2018-07-01:
It's hard to maintain the older platforms, however I didn't drop builds for
anything intentionally and the docs will stay up to date.

This was a difficult release as the ecosystem is a  little broken at the
moment. httpsthub.com/node-serialport/node-serialport/issues/1584

The previous prebuild all we we're doing no longer works, so now we're
using prebuild-ci however there is no longer a ci for all older versions of
node as our testing and linting doesn't work on them.

Happy to take a PR, but I'll try to look into it.

monkbroc, 2018-07-02:
Thank you for the quick response @reconbot. I'm sorry to hear the latest release was a pain. Tooling changes are an annoying source of churn. I made a PR to try to bring back the prebuild on older Node versions. However Node 5 still fails on Appveyor (see PR for details).

reconbot, 2018-07-02:
Oh that's great! I'll take a look.

reconbot, 2018-07-21:
#1602 fixes this

closed 2018-07-21 by reconbot