serialport/node-serialport #1591: Open (SetCommState): The parameter is incorrect

ThalesP, 2018-07-03:
- "serialport": "^6.2.0"
- __NodeJS Version__: 8.9.4
- __Operating System__ and __Hardware Platform__: Windows 10 / STM32F

### Summary of Problem
I am getting the following error when I try to open a port to communicate with a STM32F microcontroller through usb.
[image]

### Steps and Code to Reproduce the Issue
Here is the code that sets the parameters and tries to open the port.
[image]

I'm able to communicate with the usb device using a serial terminal communication software such as Termite and hercules. However, when I try to use my nodeJs application with node-serialport I get the mentioned error on Windows 10, but it works fine on Windows 7.

Does anybody know a work around this issues?

Thanks in advance.

abeck70, 2018-08-12:
I have this same issue.

abeck70, 2018-08-12:
Ok here is how to reproduce. On your USB device, make the mistake of setting the databits to 0 and say parity to 8. I.e. its not a supportable setup. For this badly configured device, windows serialport detects it in the get line encoding and then fails as above. Once I had corrected the device's default line encoding, it just works. OSX was not sensitive to this, just windows.

reconbot, 2018-09-15:
Good debugging @abeck70 thank you.

closed 2018-09-15 by reconbot

PhiInnovations, 2019-02-27:
I had the same issue and it was solved according to the following link:

https://community.st.com/s/question/0D50X00009XkfnjSAB/problem-solved-with-usb-cdc