serialport/node-serialport #1593: Feature request: Add an option to disable buffering writes to a closed port

maxwellhadley, 2018-07-05:
- __SerialPort Version__: 6.2.0
- __NodeJS Version__: N/A
- __Operating System__ and __Hardware Platform__: N/A

- Have you checked the right __version__ of the __api docs?__: N/A
- Are you having trouble installing and you checked the __Installation Special Cases__ docs? N/A
- Are you using __Electron__ and have you checked the _Electron Docs_?: N/A

I have a device which is only intermittently connected, and requires a specific handshake sequence each time it is connected. The user may disconnect it without warning. Although I now test the isOpen property before every write, there is a race condition where a device disconnect occurs between the property access and the write, which results in the write being buffered. The next time the device is connected, the handshake sequence is not correct, because of the buffered write.

This could be avoided by adding a Serialport constructor option to disable buffering of writes to a closed port.

reconbot, 2018-07-05:
Seems reasonable

maxwellhadley, 2020-07-08:
Any chance of looking at this now, after the great re-organisation?