serialport/node-serialport #1595: dependency prompt-list (prompt-radio) break node.js v4 support

altjz, 2018-07-09:
- __SerialPort Version__:
since 6.0.3 to 6.2.1
- __NodeJS Version__:
v4.6.4
- __Operating System__ and __Hardware Platform__:
Windows XP SP3

- Have you checked the right __version__ of the __api docs?__:
- Are you having trouble installing and you checked the __Installation Special Cases__ docs?
Yes, Documents mention support 4.x
- Are you using __Electron__ and have you checked the _Electron Docs_?:
No.

### Summary of Problem
Dependency prompt-list (prompt-radio) break node.js v4 support.
Here is the package.json
```json
{
  "name": "test",
  "version": "1.0.0",
  "dependencies": {
    "serialport": "^6.2.1"
  }
}

```
```
$yarn install
yarn install v1.3.2
warning You are using Node "4.4.6" which is not supported and may encounter bugs or unexpected behavior. Yarn supports the following semver range: "^4.8.0 || ^5.7.0 || ^6.2.2 || >=8.0.0"
warning package.json: No license field
info No lockfile found.
warning codemao-hardware-agent@1.2.0: No license field
[1/5] Resolving packages...
[2/5] Fetching packages...
error prompt-radio@1.2.1: The engine "node" is incompatible with this module. Expected version ">=5.0".
error Found incompatible module
info Visit https://yarnpkg.com/en/docs/cli/install for documentation about this command.
```
### Steps and Code to Reproduce the Issue
switch to node.js v4.6.4, install serialport

reconbot, 2018-07-09:
If you want to update the engines field to reflect node 4.8 I'll happily merge. 4x went through some api changes that would have waited for a new version if they were made today.

altjz, 2018-07-10:
Sure @reconbot , Glad to update the node if this could fix the issue.
I also tried v5.12.0, but still no luck.
I am just wondering why [serialport@6](https://github.com/node-serialport/node-serialport#platform-support) doesn't support node v5.x but [serialport@4](https://github.com/node-serialport/node-serialport/blob/4.0.7/README.md#platform-support) does.
I have to make my application compatible with XP, and v5.12.0 is the final version in XP.

reconbot, 2018-07-10:
It should work with 4, our tests pass with the latest nodejs 4x.

reconbot, 2018-07-21:
Works with node 4.0.0, I don't know what to say

closed 2018-07-21 by reconbot