serialport/node-serialport #1601: update deprecated Buffer usage

omnidan, 2018-07-20:
- __SerialPort Version__: 6.2.1
- __NodeJS Version__: 10.6.0
- __Operating System__ and __Hardware Platform__: macOS 10.13.6

### Summary of Problem

The following warning happens on latest node.js and serialport versions:

```
(node:5814) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
```

reconbot, 2018-07-21:
We have a call to `new Buffer()` in a test, no where else I can find. Is there a line number?

closed 2018-07-21 by reconbot

omnidan, 2018-07-21:
@reconbot it didn't give me a line number, but I will reopen this if it happens again in the next update. 😁

PR #1603 (2018-07-21): fix: Remove a new Buffer() call
closes #1601