serialport/node-serialport #1605: Is there a way to disabled DTR before openning a port

AlexandreRouma, 2018-07-23:
Hi, I'm trying to connect to an arduino without causing it to reset.
The arduino resets when DTR is set to enable. The problem is that whenever you open a port, the lib enables DTR and messes everything up :/

Hope you can help, thanks!

reconbot, 2018-07-23:
I might be able to to be proven wrong but I think this is an os thing.

reconbot, 2018-07-23:
I also recall you can disable the drt>reset link physically by cutting a
wire on the board.

AlexandreRouma, 2018-07-23:
I could cut the trace but then it gets annoying to program the arduino :/
And no, it also happens linux. It's written in the description for the port;set() function that DTR is on by default: https://node-serialport.github.io/node-serialport/SerialPort.html#set

Thanks for responding :)

kjhe11e, 2018-08-16:
@AlexandreRouma I also have a device that resets when DTR is enabled. The following worked for me:

    this.serial.on("open", function () {
        console.log("Serial port opened to device");
        self.serial.set({dtr: false});
    });

AlexandreRouma, 2018-08-16:
I tried that but sadly, it still sends a short pulse which resets the MCU :/

Tuhis, 2018-08-17:
Hi,

at least on windows platform following undocumented option works for me with Arduino.
```
    this.port = new SerialPort(this.serialportName, {
            baudRate: 250000,
            **hupcl: false,**
            dataBits: 8,
            stopBits: 1,
            parity: "none"
        });
```

AlexandreRouma, 2018-08-19:
Thanks! Gonna try that. I hope they can bring it to linux cause it's suppose to run on an RPi, but that should be enough for testing!

tablatronix, 2018-08-19:
@kjhe11e tried this on osx, didn't work, causes no data to emit

tablatronix, 2018-10-18:
Why are labelled issues marked stale, kind of lame

reconbot, 2018-10-19:
There are a high number of abandoned support issues from people who never finished debugging, which is also kind of lame.

lefuturiste, 2020-03-16:
Hi, on my raspberry pi, I managed to disable DTR, I used the command `stty -F /dev/ttyUSB0 -hupcl` before running my Node.js script. It work great once, but when I run my Node.js script a second time, the DTR seem to be disabled like before.

kjhe11e, 2020-03-16:
@lefuturiste what is your goal? Do you want DTR enabled on the second run of the script?

lefuturiste, 2020-03-17:
@kjhe11e I want DTR to be enabled at every run of the script.