serialport/node-serialport #1606: Serialport Monorepo

reconbot, 2018-07-24:
I want to break out the bindings, terminal and other components into their own packages. I also want to fold the serialport-utils repo into this one as maintaining both isn't necessary. serialport-utils did prove out the lerna project structure for our parsers.

I propose the following features and packages
- [ ] A single documentation site
- [x] `serialport` will always have everything and be very friendly users of this package shouldn't notice a change
- [x] `@serialport/parser-*` moved to this repo (maintaining history from serialport-utils as sordid as it might be since it used to be part of this repo)
- [ ] `@serialport/stream` becomes the `serialport.js` stream interface that is most of this project
- [ ] `@serialport/binding-*` become the platform specific binding packages (the challenge here is to figure out how to build individual bundled binaires and the auto download *all os* one)
- [ ] `@serialport/term` becomes the terminal cli tool
- [ ] `@serialport/list` becomes the serialport list cli tool

mathieulj, 2018-07-26:
Just to chime in that I do see a use to this as I was just poking around exactly to see if I could pull in the API without all the dependencies for the CLI.

reconbot, 2018-08-26:
I got it working, but binary builds are ... gonna have to be figured out, this is going to work differently as prebuild expects different tags.

reconbot, 2018-08-28:
I built the needed features into prebuild to make it work with lerna style tags. Should be able to have this out once the PRs land. https://github.com/prebuild/prebuild/issues/229

reconbot, 2018-08-29:
Builds aren't happening on tag push because https://developer.github.com/v3/activity/events/types/#createevent
> Note: webhooks will not receive this event for created repositories. Additionally, webhooks will not receive this event for tags if more than three tags are pushed at once.

There's no guarantee there wont be more than packages published.

reconbot, 2018-08-29:
But when I do push less than or equal to 3 tags builds work great!

PR #1664 (2018-09-15): feat: move cli tools to their own packages
- update cli docs
- fix #1659 close / disconnect error for serialport-terminal
- rename serialport-term to serialport-terminal

closes #1606
closed 2018-09-15 by reconbot