serialport/node-serialport #1609: Make a parser to pull data out from the middle of a known binary packet

reconbot, 2018-07-24:
We don't have a parser that makes this [very common problem](https://github.com/node-serialport/node-serialport/issues/1506#issuecomment-371834751) easy.

Holger-Will, 2018-12-27:
i just wrote my first parser for a known protocol. i would be willing to help. 
In my case i parse enocean packets. https://github.com/enocean-js/enocean-js/blob/master/packages/node_modules/%40enocean-js/serialport-parser/src/serialport-parser.js

the packets come in starting with a sync byte, then a header which contains information about the length of the current packet and the location of the checksums.

i guess that this is probably most common packet design. What do you have in mind?

Holger-Will, 2019-01-10:
i just followed the reference implementation from the spec ;-) ...

would be nice to have a collection of some data packages, to generalize the algorithm in a useful way, and to have some data to test against. 

I do not quite understand the packet structure from the issue you linked above. the author mentions that the packet length varies but it's not clear how to detect the different lengths.

wendy2702, 2019-04-02:
Hi,

sorry for late reply.

Please let me know how i can help?

Holger-Will, 2019-04-02:
do you have example packages of all different length and possibly information about the packet structure?

wendy2702, 2019-04-02:
Sorry i actually do not have examples for all different packages.

Do you speak German? Because the information about the packet structure is written in German.

Holger-Will, 2019-04-02:
ja, deutsch ist geht! wenn Du mir die Info zukommen lassen könntest wäre das toll ;-)

wendy2702, 2019-04-02:
das Protokoll hier hier sehr gut beschrieben:
https://www.symcon.de/forum/threads/2092-ComPort-und-Waterkotte-abfragen

Die Adressen sind hier aufgeführt:
http://www.haustechnikdialog.de/Forum/t/6144/Waterkotte-5017-3-an-den-Computer-anschliessen?page=2

Und hier auch noch eine Beschreibung:
https://www.linux-magazin.de/ausgaben/2010/06/selektive-waerme/

Leider liefert der Hersteller keine gesammelten Informationen so das man sich das zusammen suchen musste.