serialport/node-serialport #1613: StopBits support for 1.5

SVSagi, 2018-07-30:
### Summary of Problem
Not able to open port with stopbits as 1.5

I am currently building a desktop client using electron, where the COM port receives data correctly for values: baudRate: 2400, dataBits: 7, stopBits: 1.5, parity: 'none' 

I found from the node-serialport documentation, stopBits **Must be one of these: 1 or 2**. But, should 1.5 be supported as well? especially for old & slow hardware.

I assume stopbits 1.5 is a valid value(https://en.wikipedia.org/wiki/Serial_port#Stop_bits)

I am still able to read data, when stopbits 1 is used, but is not as consistent when tested with 1.5(tested with pyserial)

### Steps and Code to Reproduce the Issue
```
port = new SerialPort("COM1", { baudRate: 2400, dataBits: 7, stopBits: 1.5, parity: 'none'   }, function (err) 
    {
        if(err) 
        {
          console.log(err);//Error: Invalid stop bits setting 2
        }
    });
```

### Package name(s) and version(s)
"serialport": "^6.2.2"
"electron": "~2.0.5"

### Node version, Operating System and hardware
Node version: v8.11.3; Operating System: Ubuntu 18.04, Processor: Intel i7

### Summary of Problem
When I pass stopbits values as 1.5 with the following options, to open serial port, I get the error: 

> Error: Invalid stop bits setting 2

`{ baudRate: "2400", dataBits: "7", stopBits: "1.5", parity: 'none'   }
`
### Steps and Code to Reproduce the Issue
Create a nodejs/electron application and open a serialport connection with stopbits value as 1.5.

faraz16iqbal, 2021-01-26:
Is this still open?

reconbot, 2022-02-03:
yes, superseded by #2400