serialport/node-serialport #1616: Rpi3 based Alpine can't build serialport

rjbaat, 2018-08-03:
I had to rebuild my docker image on rpi with resin.io and i run into a problem that i can't figure out.
It won't build serialport on the rpi on alpine linux image from resin:

https://github.com/resin-io-library/base-images/blob/5159250f897aa4e88c701d4c27dac62645bafcaa/node/raspberry-pi/alpine/0.10/Dockerfile

i tried both separate node-serialport as node-red-contrib-modbus. with the command:

npm install -g node-red-contrib-modbus --production --silent
npm install -g node-red-node-serialport --production --silent

Both giving me the problem with building serialport. Any idea what the problem would be? 

```
34m[main][39m  Step 7/9 : RUN apk add --no-cache make gcc g++ python &&   JOBS=MAX npm install -g node-red-contrib-modbus --production --silent &&   apk del make gcc g++ python &&   npm cache clean --force && rm -rf /tmp/*
[34m[main][39m   ---> Running in 25b061d23a60
[34m[main][39m  fetch http://dl-cdn.alpinelinux.org/alpine/v3.8/main/armhf/APKINDEX.tar.gz
[34m[main][39m  fetch http://dl-cdn.alpinelinux.org/alpine/v3.8/community/armhf/APKINDEX.tar.gz
[34m[main][39m  (1/13) Installing binutils (2.30-r5)
[34m[main][39m  (2/13) Installing isl (0.18-r0)
[34m[main][39m  (3/13) Installing libgomp (6.4.0-r8)
[34m[main][39m  (4/13) Installing libatomic (6.4.0-r8)
[34m[main][39m  (5/13) Installing mpfr3 (3.1.5-r1)
[34m[main][39m  (6/13) Installing mpc1 (1.0.3-r1)
[34m[main][39m  (7/13) Installing gcc (6.4.0-r8)
[34m[main][39m  (8/13) Installing musl-dev (1.1.19-r10)
[34m[main][39m  (9/13) Installing libc-dev (0.7.1-r0)
[34m[main][39m  (10/13) Installing g++ (6.4.0-r8)
[34m[main][39m  (11/13) Installing make (4.2.1-r2)
[34m[main][39m  (12/13) Installing gdbm (1.13-r1)
[34m[main][39m  (13/13) Installing python2 (2.7.15-r0)
[34m[main][39m  Executing busybox-1.28.4-r0.trigger
[34m[main][39m  OK: 205 MiB in 83 packages
[34m[main][39m  make: Entering directory '/usr/local/lib/node_modules/node-red-contrib-modbus/node_modules/serialport/build'
[34m[main][39m    CXX(target) Release/obj.target/serialport/src/serialport.o
[34m[main][39m    CXX(target) Release/obj.target/serialport/src/serialport_unix.o
[34m[main][39m    CXX(target) Release/obj.target/serialport/src/poller.o
[34m[main][39m    CXX(target) Release/obj.target/serialport/src/serialport_linux.o
[34m[main][39m  [91m../src/serialport_linux.cpp:4:24: fatal error: asm/ioctls.h: No such file or directory
[34m[main][39m   #include <asm/ioctls.h>
[34m[main][39m                          ^
[34m[main][39m  compilation terminated.
[34m[main][39m  [0m
[34m[main][39m  [91mmake: *** [serialport.target.mk:96: Release/obj.target/serialport/src/serialport_linux.o] Error 1
[34m[main][39m  make: *** Waiting for unfinished jobs....
[34m[main][39m  [0m
[34m[main][39m  [91m../src/serialport_unix.cpp:26:26: fatal error: linux/serial.h: No such file or directory
[34m[main][39m   #include <linux/serial.h>
[34m[main][39m                            ^
[34m[main][39m  compilation terminated.
[34m[main][39m  [0m
[34m[main][39m  [91mmake: *** [serialport.target.mk:96: Release/obj.target/serialport/src/serialport_unix.o] Error 1
[34m[main][39m  [0m
[34m[main][39m  [91m../src/poller.cpp: In static member function 'static void Poller::onData(uv_poll_t*, int, int)':
[34m[main][39m  ../src/poller.cpp:69:29: warning: 'v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const' is deprecated [-Wdeprecated-declarations]
[34m[main][39m     obj->callback.Call(2, argv);
[34m[main][39m                               ^
[34m[main][39m  In file included from ../src/poller.cpp:1:0:
[34m[main][39m  ../../nan/nan.h:1618:3: note: declared here
[34m[main][39m     Call(int argc, v8::Local<v8::Value> argv[]) const {
[34m[main][39m     ^~~~
[34m[main][39m  [0m
[34m[main][39m  [91m../src/serialport.cpp: In function 'void EIO_AfterOpen(uv_work_t*)':
[34m[main][39m  ../src/serialport.cpp:95:30: warning: 'v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const' is deprecated [-Wdeprecated-declarations]
[34m[main][39m     data->callback.Call(2, argv);
[34m[main][39m                                ^
[34m[main][39m  In file included from ../src/./serialport.h:6:0,
[34m[main][39m                   from ../src/serialport.cpp:1:
[34m[main][39m  ../../nan/nan.h:1618:3: note: declared here
[34m[main][39m     Call(int argc, v8::Local<v8::Value> argv[]) const {
[34m[main][39m     ^~~~
[... 282 lines cut ...]
[34m[main][39m     Call(int argc, v8::Local<v8::Value> argv[]) const {
[34m[main][39m     ^~~~
[34m[main][39m  [0m
[34m[main][39m  [91m../src/serialport.cpp: In function 'void EIO_AfterDrain(uv_work_t*)':
[34m[main][39m  ../src/serialport.cpp:424:30: warning: 'v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const' is deprecated [-Wdeprecated-declarations]
[34m[main][39m     data->callback.Call(1, argv);
[34m[main][39m                                ^
[34m[main][39m  In file included from ../src/./serialport.h:6:0,
[34m[main][39m                   from ../src/serialport.cpp:1:
[34m[main][39m  ../node_modules/nan/nan.h:1618:3: note: declared here
[34m[main][39m     Call(int argc, v8::Local<v8::Value> argv[]) const {
[34m[main][39m     ^~~~
[34m[main][39m  [0m
[34m[main][39m  [91mmake: *** [serialport.target.mk:91: Release/obj.target/serialport/src/poller.o] Error 1
[34m[main][39m  [0m
[34m[main][39m  make: Leaving directory '/usr/local/lib/node_modules/node-red-node-serialport/node_modules/serialport/build'
[34m[main][39m  Removing intermediate container 47854f1ebb21
[34m[main][39m  [31mThe command '/bin/sh -c apk add --no-cache make gcc g++ python &&   JOBS=MAX npm install -g node-red-node-serialport --production --silent &&   JOBS=MAX npm install -g node-red-contrib-modbus --production --silent &&   apk del make gcc g++ python &&   npm cache clean --force && rm -rf /tmp/*' returned a non-zero code: 1[39m
[36m[Info][39m  Uploading images
[================>                                  ] 33%[2K[G[32m[Success][39m  Successfully uploaded images
[31m[Error][39m  Some services failed to build:
[31m[Error][39m    Service: main
[31m[Error][39m      Error: The command '/bin/sh -c apk add --no-cache make gcc g++ python &&   JOBS=MAX npm install -g node-red-node-serialport --production --silent &&   JOBS=MAX npm install -g node-red-contrib-modbus --production --silent &&   apk del make gcc g++ python &&   npm cache clean --force && rm -rf /tmp/*' returned a non-zero code: 1
[31m[Error][39m  Not deploying release.
```

rjbaat, 2018-08-03:
I have tried it with the comments for alpine on:

https://github.com/node-serialport/node-serialport/tree/master/packages/serialport#alpine-linux

But still getting errors and build failed.

```
[34m[main][39m  Step 7/9 : RUN apk add --no-cache make gcc g++ python linux-headers udev &&   JOBS=MAX npm install -g node-red-node-serialport --unsafe-perm --build-from-source --production --silent &&   JOBS=MAX npm install -g node-red-contrib-modbus --unsafe-perm --build-from-source --production --silent &&   apk del make gcc g++ python &&   npm cache clean --force && rm -rf /tmp/*
[34m[main][39m   ---> Running in f8029267be99
[34m[main][39m  fetch http://dl-cdn.alpinelinux.org/alpine/v3.8/main/armhf/APKINDEX.tar.gz
[34m[main][39m  fetch http://dl-cdn.alpinelinux.org/alpine/v3.8/community/armhf/APKINDEX.tar.gz
[34m[main][39m  (1/14) Installing binutils (2.30-r5)
[34m[main][39m  (2/14) Installing isl (0.18-r0)
[34m[main][39m  (3/14) Installing libgomp (6.4.0-r8)
[34m[main][39m  (4/14) Installing libatomic (6.4.0-r8)
[34m[main][39m  (5/14) Installing mpfr3 (3.1.5-r1)
[34m[main][39m  (6/14) Installing mpc1 (1.0.3-r1)
[34m[main][39m  (7/14) Installing gcc (6.4.0-r8)
[34m[main][39m  (8/14) Installing musl-dev (1.1.19-r10)
[34m[main][39m  (9/14) Installing libc-dev (0.7.1-r0)
[34m[main][39m  (10/14) Installing g++ (6.4.0-r8)
[34m[main][39m  (11/14) Installing linux-headers (4.4.6-r2)
[34m[main][39m  (12/14) Installing make (4.2.1-r2)
[34m[main][39m  (13/14) Installing gdbm (1.13-r1)
[34m[main][39m  (14/14) Installing python2 (2.7.15-r0)
[34m[main][39m  Executing busybox-1.28.4-r0.trigger
[34m[main][39m  OK: 208 MiB in 84 packages
[34m[main][39m  [91m
[34m[main][39m  [0m
[34m[main][39m  [91m/usr/local/lib/node_modules/node-red-node-serialport/node_modules/serialport/node_modules/prebuild-install/node_modules/which-pm-runs/index.js:11
[34m[main][39m  [0m
[34m[main][39m  [91m  const pmSpec = userAgent.split(' ')[0]
[34m[main][39m    ^^^^^
[34m[main][39m  [0m
[34m[main][39m  [91mSyntaxError: Use of const in strict mode.
[34m[main][39m      at Module._compile (module.js:439:25)
[34m[main][39m      at Object.Module._extensions..js (module.js:474:10)
[34m[main][39m      at Module.load (module.js:356:32)
[34m[main][39m      at Function.Module._load (module.js:312:12)
[34m[main][39m      at Module.require (module.js:364:17)
[34m[main][39m      at require (module.js:380:17)
[34m[main][39m      at Object.<anonymous> (/usr/local/lib/node_modules/node-red-node-serialport/node_modules/serialport/node_modules/prebuild-install/bin.js:5:19)
[34m[main][39m      at Module._compile (module.js:456:26)
[34m[main][39m      at Object.Module._extensions..js (module.js:474:10)
[34m[main][39m      at Module.load (module.js:356:32)
[34m[main][39m  [0m
[34m[main][39m  make: Entering directory '/usr/local/lib/node_modules/node-red-node-serialport/node_modules/serialport/build'
[34m[main][39m    CXX(target) Release/obj.target/serialport/src/serialport.o
[34m[main][39m    CXX(target) Release/obj.target/serialport/src/serialport_unix.o
[34m[main][39m    CXX(target) Release/obj.target/serialport/src/poller.o
[34m[main][39m    CXX(target) Release/obj.target/serialport/src/serialport_linux.o
[34m[main][39m  [91m../src/poller.cpp: In constructor 'Poller::Poller(int)':
[34m[main][39m  ../src/poller.cpp:12:39: error: could not convert 'status' from 'int' to 'uv_err_t {aka uv_err_s}'
[34m[main][39m       Nan::ThrowError(uv_strerror(status));
[34m[main][39m                                         ^
[34m[main][39m  [0m
[34m[main][39m  [91m../src/poller.cpp: In member function 'void Poller::poll(int)':
[34m[main][39m  ../src/poller.cpp:39:43: error: could not convert 'status' from 'int' to 'uv_err_t {aka uv_err_s}'
[34m[main][39m       Nan::ThrowTypeError(uv_strerror(status));
[34m[main][39m                                             ^
[34m[main][39m  [0m
[34m[main][39m  [91m../src/poller.cpp: In member function 'void Poller::stop()':
[34m[main][39m  ../src/poller.cpp:47:43: error: could not convert 'status' from 'int' to 'uv_err_t {aka uv_err_s}'
[34m[main][39m       Nan::ThrowTypeError(uv_strerror(status));
[34m[main][39m                                             ^
[34m[main][39m  [0m
[34m[main][39m  [91m../src/poller.cpp: In static member function 'static void Poller::onData(uv_poll_t*, int, int)':
[34m[main][39m  ../src/poller.cpp:58:75: error: could not convert 'status' from 'int' to 'uv_err_t {aka uv_err_s}'
[34m[main][39m       argv[0] = v8::Exception::Error(Nan::New<v8::String>(uv_strerror(status)).ToLocalChecked());
[34m[main][39m                                                                             ^
[... 77 lines cut ...]
[34m[main][39m     ^~~~
[34m[main][39m  [0m
[34m[main][39m  [91m../src/serialport.cpp: In function 'void EIO_AfterDrain(uv_work_t*)':
[34m[main][39m  ../src/serialport.cpp:424:30: warning: 'v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const' is deprecated [-Wdeprecated-declarations]
[34m[main][39m     data->callback.Call(1, argv);
[34m[main][39m                                ^
[34m[main][39m  In file included from ../src/./serialport.h:6:0,
[34m[main][39m                   from ../src/serialport.cpp:1:
[34m[main][39m  ../node_modules/nan/nan.h:1618:3: note: declared here
[... cut ...]
[31m[Error][39m    Service: main
[31m[Error][39m      Error: The command '/bin/sh -c apk add --no-cache make gcc g++ python linux-headers udev &&   JOBS=MAX npm install -g node-red-node-serialport --unsafe-perm --build-from-source --production --silent &&   JOBS=MAX npm install -g node-red-contrib-modbus --unsafe-perm --build-from-source --production --silent &&   apk del make gcc g++ python &&   npm cache clean --force && rm -rf /tmp/*' returned a non-zero code: 1
[31m[Error][39m  Not deploying release.
```

rjbaat, 2018-08-03:
As suggested by Resin engineer i used another Alpine base img: 8.11.3 what was a LTS version and now it does build successfully again! 👍

reconbot, 2018-08-03:
That's great!

closed 2018-08-29 by reconbot