serialport/node-serialport #1619: Serialport stops writing to encoder after some time (~> 6  hours)

lkishfy, 2018-08-07:
### Summary of Problem
Serialport stops writing to encoder after some time (~> 6  hours). This closed issue here sounds similar: https://github.com/node-serialport/node-serialport/issues/1188 , but is a bit dated -- wondering if this has been fixed in a later version of Serialport, or if I am not using Serialport correctly. I recently added `serialPort.drain(function(err,results){});`  after emitting the data to the client side, and I am testing now.

### Steps and Code to Reproduce the Issue

Code to reproduce issue below:

```
///////////////////////////////////////////////////SERIAL PORT SETUP
var io = require('socket.io')(server, { wsEngine: 'ws' });

const serialPort = new SerialPort('COM5', {
  baudRate:9600
});

//Open Encoder
serialPort.write('!', function(err) {
  if (err) {
    return console.log('Error on write: ', err.message);
  }
});

//Zero Encoder
serialPort.write('z', function(err) {
  if (err) {
    return console.log('Error on write: ', err.message);
  }
  console.log('Encoder set to zero');
});
///////////////////////////////////////////////////SERIAL PORT SETUP

///////////////////////////////////////////////////EMIT SERIAL DATA TO CLIENT
serialPort.on('data', (data) => {
  var encoderData = data.toString();
  var splitEncoderData = encoderData.split(':',3);
  var position = splitEncoderData[0];
  var direction = position - lastPosition;
  var speed = Math.abs((direction > 20) ? 20 : direction);
  console.log("Position is:" + position);

  io.sockets.emit('position', position, speed, direction);
  serialPort.drain(function(err,results){});
});
///////////////////////////////////////////////////EMIT SERIAL DATA TO CLIENT
```

### Package name(s) and version(s)
Serialport 6.1.1

### Node version, Operating System and hardware
Node 8.11.1, Windows 10

reconbot, 2018-08-29:
6 hours delayed? That seems.. like something is stuck. Do you have any information about memory usage or cpu? The code looks in good shape.

lkishfy, 2018-08-30:
Yeah, neither metric stuck out as being an issue. We ended up writing a powershell script that read the serial port and sent it to Node as a UDP datagram instead of using serialport. 

A bit puzzled on this one. I also tried re-opening and closing the port on intervals but I couldn't get that to work consistently either. @reconbot is it appropriate to use this module to read/send real-time data for extended periods of time?

reconbot, 2018-08-30:
yes it is
---
Francis Gulotta
wizard@roborooter.com

reconbot, 2018-09-15:
If you can run with debugging enabled we might get a better idea of what's going on. I don't like the idea of possible stability issues.

https://serialport.io/docs/guide-debugging#windows-command-prompt-notes

closed 2018-11-22 by stale[bot]