serialport/node-serialport #1620: Access Poller directly

PTS93, 2018-08-09:
I would like to use the Poller as described in the docs here: https://node-serialport.github.io/node-serialport/Poller.html#once
I want to register for the writable event once using the `once` function but I can't seem to figure out how to properly import this class.
I tried `var Poller = require('serialport').Poller;` but that doesn't seem to return the class.

This seems like a mundane issue or am I misunderstanding something crucial about the library architecture and I shouldn't use `Poller` directly?

reconbot, 2018-08-29:
It's not currently possible. The poller is used under the hood for linux and unix reads. If you need a package like the poller I did make https://www.npmjs.com/package/uv-poll from the code in serialport.

closed 2018-08-29 by reconbot

PTS93, 2018-08-29:
Ah great, I will use that then. Thanks!