serialport/node-serialport #1623: serial.write() not working on loop statement

factoryal, 2018-08-12:
### Summary of Problem

### Steps and Code to Reproduce the Issue

### Package name(s) and version(s)

### Node version, Operating System and hardware

### Summary of Problem

### Steps and Code to Reproduce the Issue

Node.js version: v8.11.3
serialport version: 6.2.2

I'm in trouble with this code...

This program sends a letter to my Arduino Leonardo in every 0.5 seconds.
It works perfectly fine. Arduino can recieve all messages.
<pre><code>
var SerialPort = require('serialport');
var serial = new SerialPort('COM7', {baudRate: 115200});

setInterval(function(){
    console.log(serial.write('r', 'ascii'));
}, 500);
</code></pre>
Console output:
<pre><code>
true
true
true
... ("true" printed every 0.5 seconds)
</code></pre>
<br>
<br>
<br>
This is simillar code.
But my Arduino cannot recieve any message. (RX LED not flashing)
<pre><code>
var SerialPort = require('serialport');
var serial = new SerialPort('COM7', {baudRate: 115200});
var StartTime = Date.now();
while(1){
    if(Date.now() - StartTime > 500){
        StartTime = Date.now();
        console.log(serial.write('r', 'ascii'));
    }
}
</code></pre>
Console output:
<pre><code>
true
true
true
... ("true" printed every 0.5 seconds)
</code></pre>

Is anyone knows why it's not working and how to solve this problem?
Thanks.

reconbot, 2018-08-29:
[`Serialport#write()`](https://node-serialport.github.io/node-serialport/SerialPort.html#write) doesn't take an encoding, it expects a callback or nothing. So you might be hitting errors with that.

While loops block the nodejs process and no async code can run while they are running. setTimeout or setInterval is the right way to do this.

reconbot, 2018-08-29:
https://stackoverflow.com/questions/34824460/why-does-a-while-loop-block-the-event-loop might help explain

closed 2018-08-29 by reconbot