serialport/node-serialport #1625: Paraser error

fabltd, 2018-08-14:
Hi 

There seems to be some issue with the use of the parser.

Using the following example:

```
var serialport = require("serialport");
var SerialPort = serialport.SerialPort;

var port = new SerialPort('/dev/cu.usbmodemFA131', {
    baudRate: 115200,
    parser: serialport.parsers.readline("\n") 
});
```
 
Causes the following error:

```
 parser: serialport.parsers.readline("\n") 
                               ^

TypeError: serialport.parsers.readline is not a function
```
Has something changed. I cannot seem to find this documented?

JT

kaysan13, 2018-08-16:
Hi 
test that's it
parser: new sp.parsers.Readline('\r\n')

kaysan13, 2018-08-16:
delete this line 

> 
parser: new sp.parsers.Readline('\r\n')

and add this

```
var sp = require('serialport'); 
var Delimiter = require('@serialport/parser-delimiter');
var parser = new Delimiter({ delimiter: '\r\n' })
serial.pipe(parser)
     parser.on('data', (data) => {
     resolve(data.toString('utf-8'))
           })
```