serialport/node-serialport #1632: Can't build v6.2.2. on Windows 10 x64

renatop7, 2018-08-23:
### Summary of Problem

I'm trying to build node-serialport to use in my Electron + Angular application.

### Steps and Code to Reproduce the Issue

npm install serialport

### Package name(s) and version(s)

serialport 6.2.2

### Node version, Operating System and hardware

NodeJS v10.9.0
Windows 10 x64

### Summary of Problem

Stack Trace:

```
An unhandled error occurred inside electron-rebuild
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  serialport.cpp
  serialport_win.cpp
  win_delay_load_hook.cc
..\src\serialport.cpp(95): warning C4996: 'Nan::Callback::Call': was declared deprecated [C:\Users\User\Desktop\my-project\node_modules\serialport\build\serialport.vcxproj]
  C:\Users\User\Desktop\my-project\node_modules\nan\nan.h(1618): note: see declaration of 'Nan::Callback::Call'
..\src\serialport.cpp(150): warning C4996: 'Nan::Callback::Call': was declared deprecated [C:\Users\User\Desktop\my-project\node_modules\serialport\build\serialport.vcxproj]
  C:\Users\User\Desktop\my-project\node_modules\nan\nan.h(1618): note: see declaration of 'Nan::Callback::Call'
..\src\serialport.cpp(188): warning C4996: 'Nan::Callback::Call': was declared deprecated [C:\Users\User\Desktop\my-project\node_modules\serialport\build\serialport.vcxproj]
  C:\Users\User\Desktop\my-project\node_modules\nan\nan.h(1618): note: see declaration of 'Nan::Callback::Call'
..\src\serialport.cpp(231): warning C4996: 'Nan::Callback::Call': was declared deprecated [C:\Users\User\Desktop\my-project\node_modules\serialport\build\serialport.vcxproj]
  C:\Users\User\Desktop\my-project\node_modules\nan\nan.h(1618): note: see declaration of 'Nan::Callback::Call'
..\src\serialport.cpp(285): warning C4996: 'Nan::Callback::Call': was declared deprecated [C:\Users\User\Desktop\my-project\node_modules\serialport\build\serialport.vcxproj]
  C:\Users\User\Desktop\my-project\node_modules\nan\nan.h(1618): note: see declaration of 'Nan::Callback::Call'
..\src\serialport.cpp(336): warning C4996: 'Nan::Callback::Call': was declared deprecated [C:\Users\User\Desktop\my-project\node_modules\serialport\build\serialport.vcxproj]
  C:\Users\User\Desktop\my-project\node_modules\nan\nan.h(1618): note: see declaration of 'Nan::Callback::Call'
..\src\serialport.cpp(383): warning C4996: 'Nan::Callback::Call': was declared deprecated [C:\Users\User\Desktop\my-project\node_modules\serialport\build\serialport.vcxproj]
  C:\Users\User\Desktop\my-project\node_modules\nan\nan.h(1618): note: see declaration of 'Nan::Callback::Call'
..\src\serialport.cpp(424): warning C4996: 'Nan::Callback::Call': was declared deprecated [C:\Users\User\Desktop\my-project\node_modules\serialport\build\serialport.vcxproj]
  C:\Users\User\Desktop\my-project\node_modules\nan\nan.h(1618): note: see declaration of 'Nan::Callback::Call'
..\src\serialport_win.cpp(188): warning C4311: 'type cast': pointer truncation from 'HANDLE' to 'int' [C:\Users\User\Desktop\my-project\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(188): warning C4302: 'type cast': truncation from 'HANDLE' to 'int' [C:\Users\User\Desktop\my-project\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(198): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\Users\User\Desktop\my-project\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(205): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\Users\User\Desktop\my-project\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(215): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\Users\User\Desktop\my-project\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(217): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\Users\User\Desktop\my-project\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(221): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\Users\User\Desktop\my-project\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(223): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\Users\User\Desktop\my-project\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(227): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\Users\User\Desktop\my-project\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(229): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\Users\User\Desktop\my-project\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(234): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\Users\User\Desktop\my-project\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(246): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\Users\User\Desktop\my-project\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(256): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\Users\User\Desktop\my-project\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(273): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\Users\User\Desktop\my-project\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(334): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\Users\User\Desktop\my-project\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(360): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\Users\User\Desktop\my-project\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(390): warning C4996: 'Nan::Callback::Call': was declared deprecated [C:\Users\User\Desktop\my-project\node_modules\serialport\build\serialport.vcxproj]
  C:\Users\User\Desktop\my-project\node_modules\nan\nan.h(1618): note: see declaration of 'Nan::Callback::Call'
..\src\serialport_win.cpp(463): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\Users\User\Desktop\my-project\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(480): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\Users\User\Desktop\my-project\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(493): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\Users\User\Desktop\my-project\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(503): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\Users\User\Desktop\my-project\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(531): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\Users\User\Desktop\my-project\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(542): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\Users\User\Desktop\my-project\node_modules\serialport\build\serialport.vcxproj]
..\src\serialport_win.cpp(574): warning C4996: 'Nan::Callback::Call': was declared deprecated [C:\Users\User\Desktop\my-project\node_modules\serialport\build\serialport.vcxproj]
  C:\Users\User\Desktop\my-project\node_modules\nan\nan.h(1618): note: see declaration of 'Nan::Callback::Call'
..\src\serialport_win.cpp(590): warning C4312: 'type cast': conversion from 'int' to 'HANDLE' of greater size [C:\Users\User\Desktop\my-project\node_modules\serialport\build\serialport.vcxproj]
[... 79 lines cut ...]
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\User\\Desktop\\my-project\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild" "--target=2.0.7" "--arch=x64" "--dist-url=https://atom.io/download/electron" "--build-from-source"
gyp ERR! cwd C:\Users\User\Desktop\my-project\node_modules\serialport
gyp ERR! node -v v10.9.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok

Failed with exit code: 1
    at SafeSubscriber._error (C:\Users\User\Desktop\my-project\node_modules\spawn-rx\lib\src\index.js:277:84)
[... cut ...]
    at SafeSubscriber.__tryOrSetError (C:\Users\User\Desktop\my-project\node_modules\spawn-rx\node_modules\rxjs\Subscriber.js:251:16)
    at SafeSubscriber.next (C:\Users\User\Desktop\my-project\node_modules\spawn-rx\node_modules\rxjs\Subscriber.js:191:27)
npm ERR! code ELIFECYCLE
npm ERR! errno 4294967295
npm ERR! my-project@1.0.0 install: `electron-rebuild`
npm ERR! Exit status 4294967295
npm ERR!
npm ERR! Failed at the my-project@1.0.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
```

MatxBerg, 2018-08-25:
I do have the same issue, it seems that a release package is not available for 6.2.2-electron-windows-64 bit, although it is available for 6.2.1

I had to revert to 6.2.1 for my build to work again

renatop7, 2018-08-27:
@MatxBerg You are a life savior!!!!!
Thanks thanks thanks!!!

I can't believe I lost so many hours to this... 
I guess I learned something today: "Always try old builds" haha

reconbot, 2018-08-29:
We should have all the builds back for the next version. You can build it yourself but that's a different approach and windows doesn't make it easy.

closed 2018-08-29 by reconbot