serialport/node-serialport #1646: Can't install Serialport on a Raspberry Pi running Node-Red 0.19.2

podi62, 2018-09-03:
Trying to Install serialport on a Raspberry PI 2 Rev B it fails. I've checked a previous post from 2016 but unfortunately it didn't help.

System INFOS:
"Raspbian GNU/Linux 9 (stretch)"

g++ is already the newest version (4:6.3.0-4).

gcc version 6.3.0 20170516 (Raspbian 6.3.0-18+rpi1+deb9u1) 

 node -v
v8.11.4

npm -v
6.4.1

I've run both 
sudo npm install serialport
and
sudo npm install --unsafe-perm serialport

and get the following error message:

 CXX(target) Release/obj.target/bindings/src/serialport_linux.o
  SOLINK_MODULE(target) Release/obj.target/bindings.node
collect2: fatal error: ld terminated with signal 11 [Segmentation fault]
compilation terminated.
bindings.target.mk:128: recipe for target 'Release/obj.target/bindings.node' failed
make: *** [Release/obj.target/bindings.node] Error 1
make: Leaving directory '/home/pi/node_modules/@serialport/bindings/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:262:23)
gyp ERR! stack     at emitTwo (events.js:126:13)
gyp ERR! stack     at ChildProcess.emit (events.js:214:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:198:12)
gyp ERR! System Linux 4.14.62-v7+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/node_modules/@serialport/bindings
gyp ERR! node -v v8.11.4
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN pi No description
npm WARN pi No repository field.
npm WARN pi No README data
npm WARN pi No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@2.0.2 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the @serialport/bindings@2.0.2 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2018-09-03T04_39_35_669Z-debug.log

And in the log file:

5048 verbose stack Error: @serialport/bindings@2.0.2 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
5048 verbose stack Exit status 1
5048 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:16)
5048 verbose stack     at emitTwo (events.js:126:13)
5048 verbose stack     at EventEmitter.emit (events.js:214:7)
5048 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
5048 verbose stack     at emitTwo (events.js:126:13)
5048 verbose stack     at ChildProcess.emit (events.js:214:7)
5048 verbose stack     at maybeClose (internal/child_process.js:925:16)
5048 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
5049 verbose pkgid @serialport/bindings@2.0.2
5050 verbose cwd /home/pi
5051 verbose Linux 4.14.62-v7+
5052 verbose argv "/usr/bin/node" "/usr/bin/npm" "install" "--unsafe-perm" "serialport"
5053 verbose node v8.11.4
5054 verbose npm  v6.4.1
5055 error code ELIFECYCLE
5056 error errno 1
5057 error @serialport/bindings@2.0.2 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
5057 error Exit status 1
5058 error Failed at the @serialport/bindings@2.0.2 install script.
5058 error This is probably not a problem with npm. There is likely additional logging output above.
5059 verbose exit [ 1, true ]

podi62, 2018-09-06:
Any clue about this issue?

fivdi, 2018-09-15:
The error message says:
```
collect2: fatal error: ld terminated with signal 11 [Segmentation fault]
```
This means that the linker `ld` attempted to access memory it's not allowed to access. It's not very likely that this is a serialport issue.

Try flashing a new sdcard with Raspbian to see if it helps.

closed 2018-11-22 by stale[bot]