serialport/node-serialport #1648: The serial port returns a large number of bytes not received.

976032412, 2018-09-04:
### Summary of Problem

### Steps and Code to Reproduce the Issue

### Package name(s) and version(s)

### Node version, Operating System and hardware

How to set up 80 bytes in serial port？

reconbot, 2018-09-04:
I'm afraid I'm going to need more information.

976032412, 2018-09-04:
@reconbot    
for example

   The serial port returns data as

    
[image]

  Because the byte is too long.
     

    Data will not return at this time.

   I want to get data regardless of bytes.
[image]

976032412, 2018-09-04:
@reconbot   I don't care how big the data is, I want to get it.

reconbot, 2018-09-04:
Data is emitted as it is received. As soon as the data arrives, there is a data event. It will not be all the data as that takes an unknown time to arrive. We have some parsers that might help you. 

https://node-serialport.github.io/node-serialport/ByteLengthParser.html
https://node-serialport.github.io/node-serialport/ReadLineParser.html

Or you can write your own parser if you know how your data looks.

976032412, 2018-09-04:
@reconbot   No matter how big the data is, I can get it. It's just data conversion time.
                       Really?

reconbot, 2018-09-04:
How can you know when the data has finished?

reconbot, 2018-09-04:
If you need 80 bits, use bytelength

```
const SerialPort = require('serialport')
const ByteLength = require('@serialport/parser-byte-length')
const port = new SerialPort('/dev/tty-usbserial1')
const parser = port.pipe(new ByteLength({length: 80}))
parser.on('data', console.log) // will have 80 bytes per data event
```

hanfengcan, 2018-09-04:
@reconbot but, when the length is random, bytelength can't catch all the byte..

I think need bytelength and time out ..   

such  82 byte receive this time, 10 * 8bytelength and catch 2 byte from time out

976032412, 2018-09-04:
@reconbot    How many bytes can be returned by default'data'event?

[image]

hanfengcan, 2018-09-04:
@976032412  random byte length, i think

976032412, 2018-09-04:
@hanfengcan   现在只要单片机给我返回字节特别长的     就不走  data  事件  回调了 ；所以我怀疑  时设置了接收字节的长度

976032412, 2018-09-04:
@reconbot 
Now the data event callback doesn't go as long as the microcontroller returns extra long bytes to me; so I suspected that the length of the received bytes was set.

hanfengcan, 2018-09-04:
@976032412 你的parser不要设置.可能设置1个字节太快了

976032412, 2018-09-04:
@hanfengcan    您指的是？     我好像没有设置parser；      您可以截个图我看一下么？

hanfengcan, 2018-09-04:
看您第一张图, option 那里设置了 parser: serialport.parsers.bytelength(1)

hanfengcan, 2018-09-04:
@976032412 我太久没接触这个了,帮不了您了

976032412, 2018-09-04:
@reconbot   好的！谢谢，你之前遇到过这种问题么？

hanfengcan, 2018-09-04:
没有,我之前的版本不设置 parser, 然后是接受到随机长度会回调的. 自己在写个函数解析

976032412, 2018-09-04:
@hanfengcan    好吧！  谢了

reconbot, 2018-09-04:
Btw - How old is your serialport? `parser` is no longer a constructor option and is now a transform stream. The old version of serialport has many more bugs.

976032412, 2018-09-04:
@reconbot  Max 115200

closed 2018-11-10 by stale[bot]