serialport/node-serialport #1649: Logo! (help!)

reconbot, 2018-09-04:
### Summary of Problem
While [nodebots](http://nodebots.io) has a cute logo, and I plan on moving serialport to the nodebots organization. I think it's time for serialport to have it's own logo.

[image]

[Some tweets on the subject](https://twitter.com/reconbot/status/1036805581825368065)

As I work on the #1647 new doc and guide website, I'll probably be buying a domain and hopefully submitting our site to [docusaurus](https://docusaurus.io/en/users). But for now the homepage looks like this.

[image]

(The guides are shaping up quite nicely however!)

I can put a little bit of money towards any artists we might like. The only requirement is that it has a square version and is vector. It doesn't have to be 

Suggestions? Thoughts? Submissions?

monteslu, 2018-09-04:
a few ideas:
[image]
[image]
[image]
[image]

monteslu, 2018-09-04:
a couple more with the IOIOI on a db9 connector.  Thinking sounded better than it actually turned out :)
[image]
[image]

adamcavan, 2018-09-04:
Here's a slightly more humanoid version using the original and the OIO (optional I/O?) marking but I really like the idea of the DB9 connector.
[image]

adamcavan, 2018-09-04:
And a variant based on @monteslu's inspired DB9 versions:
[image]

reconbot, 2018-09-04:
I'm in love with all these DB9 connectors. It's a perfect amount of nostolgia mixed with enough differentiation that people wont confuse it with the nodebots logo.

reconbot, 2018-09-20:
The web serial Spec has a great logo

https://github.com/WICG/serial

[image]

AddoSolutions, 2019-01-08:
Love it! If it would be of help, I can vectorize and prepare the logo for general use once a final logo is decided upon

reconbot, 2019-08-20:
We kinda settled on @monteslu's 2nd one =)

closed 2019-08-20 by reconbot