serialport/node-serialport #1652: VID and PID not consistent across platforms

PTS93, 2018-09-07:
The VID and PID returned by 'SerialPort.list()' is not consistent across Linux and Windows and possibly OSX.
On Linux I get the VID and PID in lower case, on Windows I get it in upper case.
It doesn't really matter which one is used but the string should go through a toLower or toUpper before being returned to the user.

Personally I would prefer if it was always lower case on every platform.

Shiva7071007, 2018-09-10:
I am also facing the same issue. It would be good if VID and PID are in same case across the different OS.

Shiva7071007, 2018-09-13:
Can have we this issue resolved ?

reconbot, 2018-09-13:
I haven't had time to work on it, but I'll accept a patch for it.

PTS93, 2018-09-13:
Can you point me to the file where the return object of the list function gets put together? I had a hard time finding where everything comes together, I only found the platform specific functions.
I'll gladly make a PR then.

reconbot, 2018-09-13:
Each binding is responsible for a number of actions. I apologize it looks like the docs don't currently include bindings.

Binding locations here here
https://github.com/node-serialport/node-serialport/blob/master/packages/bindings/lib/linux.js#L17-L19

https://github.com/node-serialport/node-serialport/blob/master/packages/bindings/lib/darwin.js#L17-L19

https://github.com/node-serialport/node-serialport/blob/master/packages/bindings/lib/win32.js#L10-L23

They all call out to other JS functions or c++ functions. You could handle this in JS or c++

https://github.com/node-serialport/node-serialport/blob/master/packages/bindings/src/serialport_win.cpp#L605-L928

https://github.com/node-serialport/node-serialport/blob/master/packages/bindings/src/darwin_list.cpp

https://github.com/node-serialport/node-serialport/blob/master/packages/bindings/lib/linux-list.js