serialport/node-serialport #1653: not able to view docs at https://node-serialport.github.io/node-serialport/

Shiva7071007, 2018-09-10:
### Summary of Problem
Whenever i am visiting https://node-serialport.github.io/node-serialport/ to view docs, I am getting " 404
There isn't a GitHub Pages site here."
### Steps and Code to Reproduce the Issue
visit "https://node-serialport.github.io/node-serialport/"
### Package name(s) and version(s)

### Node version, Operating System and hardware

reconbot, 2018-09-10:
> Package name(s) andare at  version(s) Node version, Operating System and

The new docs are at https://SerialPort.io I apologise, GitHub doesn't let
me setup a redirect.

reconbot, 2018-09-10:
> are new docs still work in progress?

They continue to be. Notice something missing?

Shiva7071007, 2018-09-10:
I just started to use Node, and found your old way of docs useful as they also show how to use them in a code. But i know this one be good too. Thanks.

closed 2018-09-10 by Shiva7071007

reconbot, 2018-09-10:
Most have code examples (happy to add more if there's something specific
you'd like to see), I do miss the generated API docs but I'm hoping the
guides and package layout (vs class layout) makes things easier to
understand.

Shiva7071007, 2018-09-10:
I liked how well documented ( arguments even optional one) your APIs are. Its rare with custom node-modules. Specially, the way you presented to use your API both as callback and promises with example.
I am still facing some issues but that is due to my lack of knowledge in callback and promises. I am still stuck in writing codes in synchronized fashion. Nice work.