serialport/node-serialport #1660: Upgrading to the Latest Version Data is not logging

Hawiie13, 2018-09-14:
### Summary of Problem
I am using serialport v 5 my codes are working I want to upgrade to latest ver
after update to 7.0.2 my codes are not working so i re write my code like in in the screen shot but it is not logged when i open the com port in putty its written

### Steps and Code to Reproduce the Issue
[image]
### Package name(s) and version(s)
Serialport 7.0.2

### Node version, Operating System and hardware
Node 8.12, Windows 10 1709 16299.64, 8-port modem pool wavecom

reconbot, 2018-09-15:
I added windows debugging tips https://serialport.io/docs/guide-debugging#windows-command-prompt-notes

Hawiie13, 2018-09-17:
Same Result no data/error is logging in the console

reconbot, 2018-09-17:
There should be logging just for loading up the module.

Hawiie13, 2018-09-18:
[image]
Here is the Screenshot One at Bash and on in powershell

Hawiie13, 2018-09-18:
Here is the screenshot one in bash one in powershell
[image]

reconbot, 2018-09-18:
Something is very weird because there are 4 debug logging calls that should have been printed before message written

https://github.com/node-serialport/node-serialport/blob/806e4ad903433510d73edb1af85010e785a30be3/packages/stream/stream.js#L228

https://github.com/node-serialport/node-serialport/blob/806e4ad903433510d73edb1af85010e785a30be3/packages/stream/stream.js#L231-L231

https://github.com/node-serialport/node-serialport/blob/806e4ad903433510d73edb1af85010e785a30be3/packages/stream/stream.js#L315

and 

https://github.com/node-serialport/node-serialport/blob/806e4ad903433510d73edb1af85010e785a30be3/packages/stream/stream.js#L318

A note about the code; You should attach error event listener in top level scope not in the write callback.

The fact you have no output makes me think maybe you're not actually running the right version of serialport? can you do an `npm ls`?

Hawiie13, 2018-09-18:
Here is the list
sms-server@1.0.0 C:\Users\Harry\Desktop\sms-server
`-- serialport@7.0.2
  +-- @serialport/binding-mock@2.0.1
  | +-- @serialport/binding-abstract@2.0.1
  | | `-- debug@3.2.5 deduped
  | `-- debug@3.2.5 deduped
  +-- @serialport/bindings@2.0.2
  | +-- @serialport/binding-abstract@2.0.1 deduped
  | +-- @serialport/parser-readline@2.0.1 deduped
  | +-- bindings@1.3.0
  | +-- debug@3.2.5 deduped
  | +-- nan@2.11.0
  | `-- prebuild-install@5.1.0
  |   +-- detect-libc@1.0.3
  |   +-- expand-template@1.1.1
  |   +-- github-from-package@0.0.0
  |   +-- minimist@1.2.0
  |   +-- mkdirp@0.5.1
  |   | `-- minimist@0.0.8
  |   +-- node-abi@2.4.3
  |   | `-- semver@5.5.1
  |   +-- noop-logger@0.1.1
  |   +-- npmlog@4.1.2
  |   | +-- are-we-there-yet@1.1.5
  |   | | +-- delegates@1.0.0
  |   | | `-- readable-stream@2.3.6
  |   | |   +-- core-util-is@1.0.2
  |   | |   +-- inherits@2.0.3
  |   | |   +-- isarray@1.0.0
  |   | |   +-- process-nextick-args@2.0.0
  |   | |   +-- safe-buffer@5.1.2 deduped
  |   | |   +-- string_decoder@1.1.1
  |   | |   | `-- safe-buffer@5.1.2 deduped
  |   | |   `-- util-deprecate@1.0.2
  |   | +-- console-control-strings@1.1.0
  |   | +-- gauge@2.7.4
  |   | | +-- aproba@1.2.0
  |   | | +-- console-control-strings@1.1.0 deduped
  |   | | +-- has-unicode@2.0.1
  |   | | +-- object-assign@4.1.1
  |   | | +-- signal-exit@3.0.2
  |   | | +-- string-width@1.0.2
  |   | | | +-- code-point-at@1.1.0
  |   | | | +-- is-fullwidth-code-point@1.0.0
  |   | | | | `-- number-is-nan@1.0.1
  |   | | | `-- strip-ansi@3.0.1 deduped
  |   | | +-- strip-ansi@3.0.1
  |   | | | `-- ansi-regex@2.1.1
  |   | | `-- wide-align@1.1.3
  |   | |   `-- string-width@1.0.2 deduped
  |   | `-- set-blocking@2.0.0
  |   +-- os-homedir@1.0.2
  |   +-- pump@2.0.1
  |   | +-- end-of-stream@1.4.1
  |   | | `-- once@1.4.0 deduped
  |   | `-- once@1.4.0
  |   |   `-- wrappy@1.0.2
  |   +-- rc@1.2.8
  |   | +-- deep-extend@0.6.0
  |   | +-- ini@1.3.5
  |   | +-- minimist@1.2.0 deduped
  |   | `-- strip-json-comments@2.0.1
  |   +-- simple-get@2.8.1
  |   | +-- decompress-response@3.3.0
  |   | | `-- mimic-response@1.0.1
  |   | +-- once@1.4.0 deduped
  |   | `-- simple-concat@1.0.0
  |   +-- tar-fs@1.16.3
  |   | +-- chownr@1.1.1
[... 201 lines cut ...]
          | +-- strip-color@0.1.0
          | `-- window-size@1.1.1
          |   +-- define-property@1.0.0 deduped
          |   `-- is-number@3.0.0
          |     `-- kind-of@3.2.2 deduped
          `-- static-extend@0.1.2
            +-- define-property@0.2.5
            | `-- is-descriptor@0.1.6
            |   +-- is-accessor-descriptor@0.1.6
            |   | `-- kind-of@3.2.2
            |   |   `-- is-buffer@1.1.6 deduped
            |   +-- is-data-descriptor@0.1.4
            |   | `-- kind-of@3.2.2
            |   |   `-- is-buffer@1.1.6 deduped
            |   `-- kind-of@5.1.0
            `-- object-copy@0.1.0
              +-- copy-descriptor@0.1.1
              +-- define-property@0.2.5
              | `-- is-descriptor@0.1.6
              |   +-- is-accessor-descriptor@0.1.6
              |   | `-- kind-of@3.2.2 deduped
              |   +-- is-data-descriptor@0.1.4
              |   | `-- kind-of@3.2.2 deduped
              |   `-- kind-of@5.1.0
              `-- kind-of@3.2.2 deduped

Hawiie13, 2018-09-18:
`-- serialport@7.0.2
  +-- @serialport/binding-mock@2.0.1
  | +-- @serialport/binding-abstract@2.0.1
  | | `-- debug@3.2.5 deduped
  | `-- debug@3.2.5 deduped
  +-- @serialport/bindings@2.0.2
  | +-- @serialport/binding-abstract@2.0.1 deduped
  | +-- @serialport/parser-readline@2.0.1 deduped
  | +-- bindings@1.3.0
  | +-- debug@3.2.5 deduped
  | +-- nan@2.11.0
  | `-- prebuild-install@5.1.0
  |   +-- detect-libc@1.0.3
  |   +-- expand-template@1.1.1
  |   +-- github-from-package@0.0.0
  |   +-- minimist@1.2.0
  |   +-- mkdirp@0.5.1
  |   | `-- minimist@0.0.8
  |   +-- node-abi@2.4.3
  |   | `-- semver@5.5.1
  |   +-- noop-logger@0.1.1
  |   +-- npmlog@4.1.2
  |   | +-- are-we-there-yet@1.1.5
  |   | | +-- delegates@1.0.0
  |   | | `-- readable-stream@2.3.6
  |   | |   +-- core-util-is@1.0.2
  |   | |   +-- inherits@2.0.3
  |   | |   +-- isarray@1.0.0
  |   | |   +-- process-nextick-args@2.0.0
  |   | |   +-- safe-buffer@5.1.2 deduped
  |   | |   +-- string_decoder@1.1.1
  |   | |   | `-- safe-buffer@5.1.2 deduped
  |   | |   `-- util-deprecate@1.0.2
  |   | +-- console-control-strings@1.1.0
  |   | +-- gauge@2.7.4
  |   | | +-- aproba@1.2.0
  |   | | +-- console-control-strings@1.1.0 deduped
  |   | | +-- has-unicode@2.0.1
  |   | | +-- object-assign@4.1.1
  |   | | +-- signal-exit@3.0.2
  |   | | +-- string-width@1.0.2
  |   | | | +-- code-point-at@1.1.0
  |   | | | +-- is-fullwidth-code-point@1.0.0
  |   | | | | `-- number-is-nan@1.0.1
  |   | | | `-- strip-ansi@3.0.1 deduped
  |   | | +-- strip-ansi@3.0.1
  |   | | | `-- ansi-regex@2.1.1
  |   | | `-- wide-align@1.1.3
  |   | |   `-- string-width@1.0.2 deduped
  |   | `-- set-blocking@2.0.0
  |   +-- os-homedir@1.0.2
  |   +-- pump@2.0.1
  |   | +-- end-of-stream@1.4.1
  |   | | `-- once@1.4.0 deduped
  |   | `-- once@1.4.0
  |   |   `-- wrappy@1.0.2
  |   +-- rc@1.2.8
  |   | +-- deep-extend@0.6.0
  |   | +-- ini@1.3.5
  |   | +-- minimist@1.2.0 deduped
  |   | `-- strip-json-comments@2.0.1
  |   +-- simple-get@2.8.1
  |   | +-- decompress-response@3.3.0
  |   | | `-- mimic-response@1.0.1
  |   | +-- once@1.4.0 deduped
  |   | `-- simple-concat@1.0.0
  |   +-- tar-fs@1.16.3
  |   | +-- chownr@1.1.1
  |   | +-- mkdirp@0.5.1 deduped
  |   | +-- pump@1.0.3
[... 199 lines cut ...]
          | +-- strip-color@0.1.0
          | `-- window-size@1.1.1
          |   +-- define-property@1.0.0 deduped
          |   `-- is-number@3.0.0
          |     `-- kind-of@3.2.2 deduped
          `-- static-extend@0.1.2
            +-- define-property@0.2.5
            | `-- is-descriptor@0.1.6
            |   +-- is-accessor-descriptor@0.1.6
            |   | `-- kind-of@3.2.2
            |   |   `-- is-buffer@1.1.6 deduped
            |   +-- is-data-descriptor@0.1.4
            |   | `-- kind-of@3.2.2
            |   |   `-- is-buffer@1.1.6 deduped
            |   `-- kind-of@5.1.0
            `-- object-copy@0.1.0
              +-- copy-descriptor@0.1.1
              +-- define-property@0.2.5
              | `-- is-descriptor@0.1.6
              |   +-- is-accessor-descriptor@0.1.6
              |   | `-- kind-of@3.2.2 deduped
              |   +-- is-data-descriptor@0.1.4
              |   | `-- kind-of@3.2.2 deduped
              |   `-- kind-of@5.1.0
              `-- kind-of@3.2.2 deduped

Hawiie13, 2018-09-18:
debug is not a function
[image]

reconbot, 2018-09-18:
debug is internal to serialport

closed 2018-11-24 by stale[bot]