serialport/node-serialport #1668: Not able to read the data from serial port

Vishal000, 2018-09-20:
Hi,

My usecase is I am running the NodeJS code to fetch the data from the custom made shield which uses the Modbus protocol for communication. I am using serial port to send a request to the shield and get back the response from the shield. I have written the code in NODE-RED, which has a serial port and it is able to send request and receive response. I have to connect to port /dev/ttyS0. My board in Intel Galelio Board and the board is connected to the shield via arduino pins. I have written the following code to establish the communication but the message (request) is written but I do not get any response. Please help  me with this.

var serialport = require('serialport');
var portName = '/dev/ttyS0';

var port = new serialport(portName, {
    baudRate: 115200,
	dataBits: 8,
    parity: 'none',
    stopBits: 1,
    flowControl: false
});

port.on('open', onOpen);

function onOpen(){
    console.log('Open connections!');
}

function onData(data){
    console.log('on Data ' + data);
}

port.on('readable', function () {
	console.log('Read Data:', port.read());
});

var buffer = new Buffer(9);
buffer[0] = 0x01;
buffer[1] = 0x30;
buffer[2] = 0x02;
buffer[3] = 0x07;
buffer[4] = 0x00;
buffer[5] = 0x00;
buffer[6] = 0x02;
buffer[7] = 0x00;
buffer[8] = 0x00;

port.write(buffer, function(err) {
  if (err) {
    return console.log('Error on write: ', err.message);
  }
  console.log('message written');
});

port.on('data', onData);

reconbot, 2018-09-20:
You don't want to mix `port.read()` with `post.on('data', ...)` use one or the other.

Vishal000, 2018-09-21:
I get the following output when I use the debug mode.

$ DEBUG=* node flow.js
  @serialport/bindings loading LinuxBinding +0ms
  @serialport/stream opening path: /dev/ttyS0 +0ms
  @serialport/binding-abstract open +0ms
  @serialport/stream _read queueing _read for after open +189ms
  @serialport/bindings:poller Creating poller +0ms
  @serialport/stream opened path: /dev/ttyS0 +74ms
Open connections!
  @serialport/stream _write 9 bytes of data +130ms
  @serialport/binding-abstract write 9 bytes +331ms
  @serialport/stream _read reading +40ms
  @serialport/binding-abstract read +40ms
  @serialport/bindings:unixWrite Starting write 9 bytes offset 0 bytesToWrite 9 +0ms
  @serialport/bindings:unixRead Starting read +0ms
  @serialport/bindings:unixWrite write returned null 9 +50ms
  @serialport/bindings:unixWrite wrote 9 bytes +27ms
  @serialport/bindings:unixWrite Finished writing 9 bytes +2ms
  @serialport/stream binding.write write finished +105ms
message written
  @serialport/bindings:unixRead waiting for readable because of code: EAGAIN +76ms
  @serialport/bindings:poller Polling for "readable" +311ms

I am not getting any data. What do you suggest ?

reconbot, 2018-09-21:
That's telling me it wrote 9 bytes and was told by your os that there
wasn't any more data. Do you have the right baudrate?

Vishal000, 2018-09-24:
Yes the baud rate is correct. It is working in Node RED framework but not in Node JS.

Vishal000, 2018-09-25:
Is it any problem in the code or the library ?
I am trying all possible mechanisms to read data but it is not possible. Could anyone provide an insight inot the solution ?

HipsterBrown, 2018-09-25:
@Vishal000 What version of serialport are you using?

Vishal000, 2018-09-26:
Node: 6.12.3
Serialport: 6.2.2

Vishal000, 2018-09-28:
Can anyone please help with this situation ?

closed 2018-12-04 by stale[bot]