serialport/node-serialport #1671: Reading serial data from arduino on raspberry pi is breaking the data

jcalmeida, 2018-09-20:
### Summary of Problem

I'm trying to send data from an arduino uno to raspberry pi 3B using serial over USB.
Arduino is sending a simple string "Hello" but the rpi receives it like this:
Open connection
He
llo

H
ello

Hell
o

He
llo

H
ello
(...)

I would expect it to be like this:
Hello
Hello
Hello
(...)

I've searched for this issue but I don't seem to get anyone with the exact same error. Increasing baudRate seems to improve it, and sometimes I get a full "Hello", but even so it's not consistent.

It seems a parser problem, but I can confirm that the buffer returns the CR LF (0d 0a). However, the buffer packages seem to have different sizes:
<Buffer 48 65>
<Buffer 6c 6c 6f 0d 0a>
<Buffer 48>
<Buffer 65 6c 6c 6f>
<Buffer 0d 0a>
<Buffer 48 65 6c 6c>
<Buffer 6f 0d 0a>
<Buffer 48 65>
<Buffer 6c 6c 6f 0d>
<Buffer 0a>
<Buffer 48>
<Buffer 65 6c 6c 6f>

### Code to Reproduce the Issue

The nodejs Code:
```js
const SerialPort = require("serialport");

var myPort = new SerialPort ("/dev/ttyACM0", {
        baudRate: 9600,
        parser: new SerialPort.parsers.Readline("\r\n"),
})

myPort.on('open', onOpen);
myPort.on('data', onData);

function onOpen()
{
        console.log("Open connection");
;}

function onData(data)
{
        console.log(data.toString('UTF8'));
}
```

The Arduino code:
```c++
void setup() {
  Serial.begin(9600);
}

void loop() {
  Serial.println("Hello");
  delay (1000);
}
```

### Versions, Operating System and Hardware

- SerialPort@: 7.0.2
- Node.js: (not sure :(...new to this)
- Linux (raspbian)
- Raspberry pi 3B

HipsterBrown, 2018-09-21:
@jcalmeida Have you tried the `readline.js` example? https://github.com/node-serialport/node-serialport/blob/%40serialport/bindings%402.0.2/packages/serialport/examples/readline.js

jcalmeida, 2018-09-21:
> @jcalmeida Have you tried the `readline.js` example? https://github.com/node-serialport/node-serialport/blob/%40serialport/bindings%402.0.2/packages/serialport/examples/readline.js
> 
> Edited the example to match your configuration:
> 
> ```js
> // Use a Readline parser
> 
> const SerialPort = require('serialport')
> const parsers = SerialPort.parsers
> 
> // Use a `\r\n` as a line terminator
> const parser = new parsers.Readline({
>   delimiter: '\r\n',
> })
> 
> const port = new SerialPort('/dev/ttyACM0', {
>   baudRate: 9600,
> })
> 
> port.pipe(parser)
> 
> port.on('open', () => console.log('Port open'))
> 
> parser.on('data', console.log)
> ```
> 
> The `SerialPort` constructor does not take a parser option.

Simple as that. It works! Thanks a lot!
I've search for a lot of online examples, and the parser was always on the constructor. I couldn't see the constructor definition in the API page and didn't realize there was a Readline example.... Shame on me.

Thanks a lot for pointing me to it. I spent more time on a simple issue that I'd like to admit! :)

closed 2018-09-21 by jcalmeida

HipsterBrown, 2018-09-22:
No worries. We’re working to make the getting started experience better, so any feedback is helpful. Good luck on your project! :+1: