serialport/node-serialport #1672: Missing electron 3.0.0 NODE_MODULE_VERSION 64 prebuild package

WolfWalter, 2018-09-21:
## 💥 Proposal

### What feature you'd like to see
Adding a prebuild package for electron 3.0.0 which was just released.

## Motivation
I wanted to test my app with electron 3.0.0 and got an error:

```
ERROR Error: Uncaught (in promise): Error: The module '.../node_modules/@serialport/bindings/build/Release/bindings.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 64...
```

HipsterBrown, 2018-09-21:
@WolfWalter Have you tried compiling serialport for the latest version of electron locally? https://serialport.io/docs/guide-installation#electron

HipsterBrown, 2018-09-21:
@WolfWalter Oh I see what happened. We need to do another patch release to prebuild for the latest electron. Thanks for creating an issue. :+1:

HipsterBrown, 2018-09-21:
I've added a prebuilt binary for electron@3.0.0 on Mac to the current release. I'm looking into how I can build the various windows and linux versions as well.

daniel-faber, 2018-11-06:
Any news on this?

Electron 4 is already in beta 5, so we'll also need prebuilt binary for electron@4 soon.

reconbot, 2018-11-06:
You can always build your own binaries if you need something specific.
We'll support thee next electron when prebuild supports it. We're cutting a
new release in the next day with the latest prebuild.

-Francis

---
Francis Gulotta
wizard@roborooter.com

Sneezry, 2018-12-17:
Hi, VS Code has already upgraded to electron 3.0.10 in the insider version, and the change will be applied to stable channel soon. Please consider to support electron 3.0.0, or all VS Code extensions depending on this module will be broken. Thank you.

reconbot, 2018-12-17:
Don't we?

Sneezry, 2018-12-17:
Seems not. There's not prebuild package for electron 3.0: https://github.com/node-serialport/node-serialport/releases

HipsterBrown, 2018-12-17:
@Sneezry Is `v64` not the latest ABI for electron 3.0? https://github.com/node-serialport/node-serialport/releases/tag/%40serialport%2Fbindings%402.0.3

HipsterBrown, 2018-12-17:
So I updated the [electron-serialport example](
https://github.com/johnny-five-io/electron-serialpor) locally to the latest electron and serialport versions to test the prebuilt binaries. It appears to be an issue with `electron-rebuild` perhaps not getting the bindings file correctly. When I download the `bindings-2.0.3-electron-v64-darwin-x64.tar.gz` under the `@serialport/bindings@2.0.3` release and cp the `bindings.node` file to `node_modules/@serialport/bindings/build/Release/bindings.node`, the app starts and works. 

Still looking into why the prebuilts for electron are not being pulled down by `electron-rebuild` or `prebuild-install`.

closed 2018-12-19 by HipsterBrown

HipsterBrown, 2018-12-19:
This should now be fixed with the release of `serialport@7.1.1`, which includes `@serialport/bindings@2.0.4`. The prebuilt binaries are still building but using `electron-rebuild` will succeed in building the bindings for electron 3.x.x. 

Thanks @Sneezry for creating an issue under the `nan` project which lead to this fix. :tada:

Happy building everyone!

PR #1746 (2018-12-19): Update nan to 2.12.1
Replaces https://github.com/node-serialport/node-serialport/pull/1745

Fixes #1672

See [`nan` changelog](https://github.com/nodejs/nan/blob/master/CHANGELOG.md#2121-dec-18-2018) for release notes.