serialport/node-serialport #1673: Make "https://github.com/node-serialport/node-serialport#to-install" work

jenschr, 2018-09-21:
## 📚 Documentation

Upon installing the module "node-red-node-arduino", many users will get problems compiling the module and node-serialport is the cause. I just had 26 students try to install and more than half of the users on OSX had this problem, myself included. This is what the error looks like when starting node-red:

> It looks like serialport didn't compile properly. This is a common problem and its fix is well documented here https://github.com/voodootikigod/node-serialport#to-install
> 21 Sep 14:53:17 - [warn] ------------------------------------------------------
> 21 Sep 14:53:17 - [warn] [node-red-node-arduino/arduino] Missing serialport dependency
> 21 Sep 14:53:17 - [warn] ------------------------------------------------------

It's fantastic that node-red tries to solve this, but it seems the documentation has been changed so the link is no longer valid? I found a working version of this link in:
https://github.com/node-serialport/node-serialport/blob/v1.7.4/README.md#to-install
... but this is surprisingly old? Is this just legacy that should have been removed ages ago? I've had 4-5 stabs at various ways to solve this and I still haven't found the solution. A valid troubleshooting link would be great at this time.

### Have you read the [Contributing Guidelines on docs](https://github.com/node-serialport/node-serialport/blob/master/CONTRIBUTING.md#writing-documentation)?

Yes. From the looks of it, feedback on the main README.md is not covered but it should surely be as it's documentation?

HipsterBrown, 2018-09-21:
@jenschr Thanks for creating an issue. The new installation docs are located here -> https://serialport.io/docs/guide-installation

Do you know what version of Node the students were using?

jenschr, 2018-09-21:
They used somewhat differing versions, but most of them were using node 9.1.0. I'm on OSX 10.12.6 and I can't get serialport to work despite building from source. It builds fine (but complains that "'Call' has been explicitly marked deprecated").

One thing - upon starting I realise that I'm running Node-RED version: v0.14.6. That does not look very current. Do you know what versions is supported for the current release?

reconbot, 2018-09-23:
I can't find that readme link in any existing code or docs for `serialport`, `node-red` or `node-red-node-arduino`. It's possible its in an older version, but we haven't had the project under that url for almost 4 years. Its possible you need to upgrade.

closed 2018-09-23 by reconbot