serialport/node-serialport #1675: Can not receive any data from connected device.

ChrisJVoss, 2018-09-21:
Mac OS v10.13.6, Electron v2.0.8, serialport v7.0.2

Hi, I am building an Electron app which captures data from several CPWplus-75 scales connected via rs-232. 

I am connecting the scale to my computer using a usb 2.0 to serial adapter. 
Calling `SerialPort.list()` shows a port with manufacturer "FTDI", the adapter's manufacturer. 

    import SerialPort from 'serialport'
    const port = new SerialPort('/dev/tty.usbserial-A9076Y4J')

    port.on("open", () => {
      console.log('open');
      port.on('data', data => {
          console.log(data);
      });
    });

I get "open" in the console but no data is logged. The scale is configured to continuously stream data.

My initial thought is that the scale is not actually being picked up and that all I'm seeing from `SerialPort.list()` is the adapter cable. The adapter came with drivers and I suspect that maybe these drivers did not install properly?

My two main questions:
1) Are there driverless adapters for rs-232 to usb?
2) When a device is properly connected, will the device's information (manufacturer, productId, serialNumber, etc...) show up or will it only be the adapter cable's information?

HipsterBrown, 2018-09-22:
Can you check out the debugging output and reply with the logs? https://serialport.io/docs/guide-debugging#debugging-output

HipsterBrown, 2018-10-02:
@ChrisJVoss Are you still having this issue? Asking since you closed this more recent issue -> https://github.com/node-serialport/node-serialport/issues/1686

ChrisJVoss, 2018-10-03:
@HipsterBrown 
I did figure it out. I needed a null-modem cable. Thank you for your help!

closed 2018-10-03 by ChrisJVoss