serialport/node-serialport #1680: Memory leak still exist on raspberry pi Linux

behi88, 2018-09-23:
## 👉 [Please follow one of these issue templates](https://github.com/node-serialport/node-serialport/issues/new/choose) 👈

Hardware: Raspberry pi 
Serial Port : ttyUSB0
node version : 10.10

the memory leak still exist but really slowly
on  20 write command (10 byte each)  it increase memory 2.3MB in hour
and system fails in 12 - 15 days!

any ideas?

behi88, 2018-09-23:
@reconbot

reconbot, 2018-09-23:
I need code and version information.

behi88, 2018-09-24:
it is the latest version that I download yesterday
the code is :

var SerialPort = require('serialport');
var port = new SerialPort('/dev/ttyUSB0', 
{
  baudRate: 9600
});
setInterval(function() 
{
	console.log(process.memoryUsage());
	port.write("main screen turn on\n", function(err) 
	{
		if (err) 
		{
			return console.log('Error on write: ', err.message);
		}
	});
},500);

HipsterBrown, 2018-10-02:
@behi88 Can you provide more information around what OS you're running on the Raspberry Pi? 

I heard about a similar type of issue in the johnny-five repo, so maybe check that out for some additional debugging help -> https://github.com/rwaldron/johnny-five/issues/1504#issuecomment-421485450

behi88, 2018-10-06:
Hi
I use "htop" for debugging ans I use Raspbian 4.14 stretch.
I found the memory leak in node-modbus library and narrow it down to this

closed 2018-12-12 by stale[bot]