serialport/node-serialport #1681: Ocasional Bluescreen of death in windows 10

mreis1, 2018-09-23:
### Summary of Problem

- What are you trying to do?
Communicating with a payment terminal

- What happens?
A Bluescreen of death is thrown from time to time

### Code to Reproduce the Issue
Actually, this issue is hard to reproduce. My code works correctly and the last time I faced this fatal error I was receiving serial data from the slave peripheral, which makes me think that something is going really wrong internally. In a normal dev routine (8hours), I will run 1/2/3 times into this fatal error. Since the system never crashes when I'm not doing serial communication, I'm pretty sure that the cause is related to the communication. 

### Versions, Operating System and Hardware

- SerialPort@6.0.5
- Node.js v8
- Windows 10

reconbot, 2018-09-23:
What drivers or chipsets are you using? Have you tried upgrading them?

mreis1, 2018-09-23:
I'm using a prolific usb/serial converter just like [this one](http://www.tronisoft.com/shop/prolific-usb-dual-serial.html). Do you think this bluescreen might be caused by it's drivers? 

I forgot to mention an important detail. My app is running inside a Windows 10 VM (powered by Parallels). My Mac's os version is 10.13.X.

I'll try to update the prolific drivers to see how it goes.

mreis1, 2018-09-24:
Chip Version: 

[image]

Drivers Updated: 
[image]

mreis1, 2018-09-28:
I've two implementations of the same serial protocol. One of them is a C# implementation which works great - at least I never came across a blue screen of death while using it and it uses the same adapter that I'm using in my node app.
In the other-hand, certain communications occasionally throw a blue screen of death.
When I say occasional, I mean, the same command can work now, but it can crash 10 min later or 3 hours later.

What could possibly be wrong?

reconbot, 2018-10-02:
That's a good question, can you share your c# code? We use the c++ API and
not the .net apis but it would be helpful to compare.

mreis1, 2018-10-02:
Actually I don't have access to the C# code. The program was created by a third party company and it's binaries are available for demonstration purposes. Based on their documentation I did the necessary to implement the API in node but this mysterious bluescreens keep showing up. 
I updated from SerialPort 6.0.5 to 6.2.2 . I'll do some additional tests to see if those bluescreen errors are eventually gone

reconbot, 2018-10-02:
If you can somehow get us a screenshot of the error, that might help too
---
Francis Gulotta
wizard@roborooter.com

reconbot, 2018-10-04:
There must be a log of what the error really is beyond wdf violation (windows driver framework)

mreis1, 2018-10-05:
Yes, I guess I got something from my last blue screen which occurred yesterday evening - 2018-10-04 @ 18:55:59

I attached a zip containing  a crash-log.evtx
[crash-log.evtx.zip](https://github.com/node-serialport/node-serialport/files/2450169/crash-log.evtx.zip)

[image]

[image]

reconbot, 2018-10-05:
I don't have any way to read that event log on Linux or Mac (I don't have
regular access to windows development machines) but the second event
doesn't seem related, we don't use com services.

mreis1, 2018-10-17:
I've been testing the app in a pre-prod environment (no Prologic adapters and no VM's) and it looks stable enough since until now, no BOD were produced. I'll keep my tests going and If the problem is not produced, I'll close this issue.

closed 2018-12-24 by stale[bot]