serialport/node-serialport #1682: Installation issue on raspberry pi

ellisium, 2018-09-24:
### Summary of Problem

- What are you trying to do?
installation
- What happens?
installation fail
- What should have happened?

`sudo npm install serialport --unsafe-perm --build-from-source

> @serialport/bindings@2.0.2 install /home/osmc/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

gyp ERR! build error 
gyp ERR! stack Error: not found: make
gyp ERR! stack     at getNotFoundError (/home/osmc/.nvm/versions/node/v7.10.1/lib/node_modules/npm/node_modules/which/which.js:13:12)
gyp ERR! stack     at F (/home/osmc/.nvm/versions/node/v7.10.1/lib/node_modules/npm/node_modules/which/which.js:68:19)
gyp ERR! stack     at E (/home/osmc/.nvm/versions/node/v7.10.1/lib/node_modules/npm/node_modules/which/which.js:80:29)
gyp ERR! stack     at /home/osmc/.nvm/versions/node/v7.10.1/lib/node_modules/npm/node_modules/which/which.js:89:16
gyp ERR! stack     at /home/osmc/.nvm/versions/node/v7.10.1/lib/node_modules/npm/node_modules/which/node_modules/isexe/index.js:44:5
gyp ERR! stack     at /home/osmc/.nvm/versions/node/v7.10.1/lib/node_modules/npm/node_modules/which/node_modules/isexe/access.js:8:5
gyp ERR! stack     at FSReqWrap.oncomplete (fs.js:114:15)
gyp ERR! System Linux 4.14.34-4-osmc
gyp ERR! command "/home/osmc/.nvm/versions/node/v7.10.1/bin/node" "/home/osmc/.nvm/versions/node/v7.10.1/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/osmc/node_modules/@serialport/bindings
gyp ERR! node -v v7.10.1
gyp ERR! node-gyp -v v3.5.0
gyp ERR! not ok 
npm WARN enoent ENOENT: no such file or directory, open '/home/osmc/package.json'
npm WARN osmc No description
npm WARN osmc No repository field.
npm WARN osmc No README data
npm WARN osmc No license field.
npm ERR! Linux 4.14.34-4-osmc
npm ERR! argv "/home/osmc/.nvm/versions/node/v7.10.1/bin/node" "/home/osmc/.nvm/versions/node/v7.10.1/bin/npm" "install" "serialport" "--unsafe-perm" "--build-from-source"
npm ERR! node v7.10.1
npm ERR! npm  v4.2.0
npm ERR! code ELIFECYCLE
npm ERR! errno 1

npm ERR! @serialport/bindings@2.0.2 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the @serialport/bindings@2.0.2 install script 'prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the @serialport/bindings package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs @serialport/bindings
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls @serialport/bindings
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /root/.npm/_logs/2018-09-24T15_38_26_055Z-debug.log
`

### Versions, Operating System and Hardware

- SerialPort@?
 latest
- Node.js v?
 7.10.1
- Windows? Linux? Mac?
linux raspberry pi (OSMC)
- Hardware and chipset? (Prolific/FTDI/Other)

HipsterBrown, 2018-09-24:
> gyp ERR! stack Error: not found: make

Do you have `make` installed on your system?

ellisium, 2018-09-24:
Oh my bad, it's a fresh install and yes missing make and g++
thank you very much

closed 2018-09-24 by ellisium