serialport/node-serialport #1683: On Linux Mint 19: Can't use firmdata-party. Seems problem related to node-serialport

jcabrerazuniga, 2018-09-24:
Can't use firmdata-party. I installed jhonny-five before. 

npm install firmata-party -save

    serialport@4.0.7 install /home/username/Documents/arduinoPrograms/node_modules/avrgirl-arduino/node_modules/serialport
    node-pre-gyp install --fallback-to-build

node-pre-gyp info it worked if it ends with ok
node-pre-gyp verb cli [ '/usr/bin/node',
node-pre-gyp verb cli '/home/username/Documents/arduinoPrograms/node_modules/avrgirl-arduino/node_modules/serialport/node_modules/.bin/node-pre-gyp',
node-pre-gyp verb cli 'install',
node-pre-gyp verb cli '--fallback-to-build' ]
node-pre-gyp info using node-pre-gyp@0.6.32
node-pre-gyp info using node@10.11.0 | linux | x64
node-pre-gyp verb command install []
node-pre-gyp info check checked for "/home/username/Documents/arduinoPrograms/node_modules/avrgirl-arduino/node_modules/serialport/build/Release/serialport.node" (not found)
node-pre-gyp http GET https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v64-linux-x64.tar.gz
node-pre-gyp http 404 https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v64-linux-x64.tar.gz
node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v64-linux-x64.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@10.11.0 (node-v64 ABI) (falling back to source compile with node-gyp)
node-pre-gyp http 404 status code downloading tarball https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v64-linux-x64.tar.gz
node-pre-gyp verb command build [ 'rebuild' ]
node-pre-gyp ERR! Tried to download(undefined): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v64-linux-x64.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@10.11.0 (node-v64 ABI) (falling back to source compile with node-gyp)
node-pre-gyp http Connection closed while downloading tarball file
node-pre-gyp verb command build [ 'rebuild' ]
make: Entering directory '/home/username/Documents/arduinoPrograms/node_modules/avrgirl-arduino/node_modules/serialport/build'
CXX(target) Release/obj.target/serialport/src/serialport.o
make: Entering directory '/home/username/Documents/arduinoPrograms/node_modules/avrgirl-arduino/node_modules/serialport/build'
CXX(target) Release/obj.target/serialport/src/serialport.o
../src/serialport.cpp: In function ‘int getIntFromObject(v8::Localv8::Object, std::__cxx11::string)’:
../src/serialport.cpp:90:52: error: no matching function for call to ‘v8::Value::ToInt32()’
return getValueFromObject(options, key)->ToInt32()->Int32Value();
^
In file included from /home/username/.node-gyp/10.11.0/include/node/node.h:63:0,
from ../../../../nan/nan.h:52,
from ../src/./serialport.h:7,
from ../src/serialport.cpp:1:
/home/username/.node-gyp/10.11.0/include/node/v8.h:2433:43: note: candidate: v8::MaybeLocalv8::Int32 v8::Value::ToInt32(v8::Localv8::Context) const
V8_WARN_UNUSED_RESULT MaybeLocal ToInt32(Local context) const;
^~~~~~~
/home/username/.node-gyp/10.11.0/include/node/v8.h:2433:43: note: candidate expects 1 argument, 0 provided
In file included from /home/username/.node-gyp/10.11.0/include/node/v8.h:26:0,
from /home/username/.node-gyp/10.11.0/include/node/node.h:63,
from ../../../../nan/nan.h:52,
from ../src/./serialport.h:7,
from ../src/serialport.cpp:1:
/home/username/.node-gyp/10.11.0/include/node/v8.h:2446:34: note: candidate: v8::Localv8::Int32 v8::Value::ToInt32(v8::Isolate*) const
Local ToInt32(Isolate* isolate) const);
^
/home/username/.node-gyp/10.11.0/include/node/v8config.h:346:48: note: in definition of macro ‘V8_DEPRECATE_SOON’
#define V8_DEPRECATE_SOON(message, declarator) declarator
^~~~~~~~~~
/home/username/.node-gyp/10.11.0/include/node/v8.h:2446:34: note: candidate expects 1 argument, 0 provided
Local ToInt32(Isolate* isolate) const);
^
/home/username/.node-gyp/10.11.0/include/node/v8config.h:346:48: note: in definition of macro ‘V8_DEPRECATE_SOON’
#define V8_DEPRECATE_SOON(message, declarator) declarator
^~~~~~~~~~
../src/serialport.cpp: In function ‘double getDoubleFromObject(v8::Localv8::Object, std::__cxx11::string)’:
../src/serialport.cpp:102:53: error: no matching function for call to ‘v8::Value::ToNumber()’
return getValueFromObject(options, key)->ToNumber()->NumberValue();
^
In file included from /home/username/.node-gyp/10.11.0/include/node/node.h:63:0,
from ../../../../nan/nan.h:52,
from ../src/./serialport.h:7,
from ../src/serialport.cpp:1:
/home/username/.node-gyp/10.11.0/include/node/v8.h:2421:44: note: candidate: v8::MaybeLocalv8::Number v8::Value::ToNumber(v8::Localv8::Context) const
V8_WARN_UNUSED_RESULT MaybeLocal ToNumber(
^~~~~~~~
[... 739 lines cut ...]
gyp ERR! System Linux 4.15.0-20-generic
node-pre-gyp gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/home/username/Documents/arduinoPrograms/node_modules/avrgirl-arduino/node_modules/serialport/build/Release/serialport.node" "--module_name=serialport" "--module_path=/home/username/Documents/arduinoPrograms/node_modules/avrgirl-arduino/node_modules/serialport/build/Release"
ERR!gyp ERR! cwd /home/username/Documents/arduinoPrograms/node_modules/avrgirl-arduino/node_modules/serialport
gyp ERR! node -v v10.11.0
commandgyp ERR! node-gyp -v v3.8.0
"/usr/bin/node" "/home/username/Documents/arduinoPrograms/node_modules/avrgirl-arduino/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
gyp ERR! not ok
node-pre-gyp ERR! cwd /home/username/Documents/arduinoPrograms/node_modules/avrgirl-arduino/node_modules/serialport
node-pre-gyp ERR! node -v v10.11.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.32
node-pre-gyp ERR! not ok
Failed to execute '/usr/bin/node /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/username/Documents/arduinoPrograms/node_modules/avrgirl-arduino/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/home/username/Documents/arduinoPrograms/node_modules/avrgirl-arduino/node_modules/serialport/build/Release' (1)
npm WARN arduinoPrograms@1.0.0 No description
npm WARN arduinoPrograms@1.0.0 No repository field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! serialport@4.0.7 install: node-pre-gyp install --fallback-to-build
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@4.0.7 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR! /home/username/.npm/_logs/2018-09-22T18_27_19_476Z-debug.log

HipsterBrown, 2018-09-24:
Do you have the `build-essential` package installed on your system? https://serialport.io/docs/guide-installation#ubuntu-debian-linux

Are you able to run `npm install serialport@4.0.1 --build-from-source` successfully?

jcabrerazuniga, 2018-09-25:
I have the build-essential package and after trying the command you provided:

~/Documents/arduinoPrograms$ npm install serialport@4.0.1 --build-from-source

> serialport@6.2.2 install /home/username/Documents/arduinoPrograms/node_modules/johnny-five/node_modules/serialport
> prebuild-install || node-gyp rebuild

make: Entering directory '/home/username/Documents/arduinoPrograms/node_modules/johnny-five/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Open(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:41:49: warning: ‘v8::String::Utf8Value::Utf8Value(v8::Local<v8::Value>)’ is deprecated: Use Isolate version [-Wdeprecated-declarations]
   v8::String::Utf8Value path(info[0]->ToString());
                                                 ^
In file included from /home/username/.node-gyp/10.11.0/include/node/v8.h:26:0,
                 from /home/username/.node-gyp/10.11.0/include/node/node.h:63,
                 from ../../../../nan/nan.h:52,
                 from ../src/./serialport.h:6,
                 from ../src/serialport.cpp:1:
/home/username/.node-gyp/10.11.0/include/node/v8.h:2888:28: note: declared here
                   explicit Utf8Value(Local<v8::Value> obj));
                            ^
/home/username/.node-gyp/10.11.0/include/node/v8config.h:324:3: note: in definition of macro ‘V8_DEPRECATED’
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
../src/serialport.cpp: In function ‘void EIO_AfterOpen(uv_work_t*)’:
../src/serialport.cpp:95:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated [-Wdeprecated-declarations]
   data->callback.Call(2, argv);
                              ^
In file included from ../src/./serialport.h:6:0,
                 from ../src/serialport.cpp:1:
../../../../nan/nan.h:1647:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
../src/serialport.cpp: In function ‘void EIO_AfterUpdate(uv_work_t*)’:
../src/serialport.cpp:150:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated [-Wdeprecated-declarations]
   data->callback.Call(1, argv);
                              ^
In file included from ../src/./serialport.h:6:0,
                 from ../src/serialport.cpp:1:
../../../../nan/nan.h:1647:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
../src/serialport.cpp: In function ‘void EIO_AfterClose(uv_work_t*)’:
../src/serialport.cpp:188:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated [-Wdeprecated-declarations]
   data->callback.Call(1, argv);
                              ^
In file included from ../src/./serialport.h:6:0,
                 from ../src/serialport.cpp:1:
../../../../nan/nan.h:1647:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
../src/serialport.cpp: In function ‘void EIO_AfterFlush(uv_work_t*)’:
../src/serialport.cpp:231:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated [-Wdeprecated-declarations]
   data->callback.Call(1, argv);
                              ^
In file included from ../src/./serialport.h:6:0,
                 from ../src/serialport.cpp:1:
../../../../nan/nan.h:1647:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
../src/serialport.cpp: In function ‘void EIO_AfterSet(uv_work_t*)’:
../src/serialport.cpp:285:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated [-Wdeprecated-declarations]
   data->callback.Call(1, argv);
                              ^
In file included from ../src/./serialport.h:6:0,
                 from ../src/serialport.cpp:1:
../../../../nan/nan.h:1647:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
../src/serialport.cpp: In function ‘void EIO_AfterGet(uv_work_t*)’:
[... 467 lines cut ...]
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/username/Documents/arduinoPrograms/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:962:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
node-pre-gyp ERR! System Linux 4.15.0-20-generic
node-pre-gyp ERR! command "/usr/bin/node" "/home/username/Documents/arduinoPrograms/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/username/Documents/arduinoPrograms/node_modules/serialport
node-pre-gyp ERR! node -v v10.11.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.29
node-pre-gyp ERR! not ok 
Failed to execute '/usr/bin/node /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/username/Documents/arduinoPrograms/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/home/username/Documents/arduinoPrograms/node_modules/serialport/build/Release' (1)
npm WARN arduinoPrograms@1.0.0 No description
npm WARN arduinoPrograms@1.0.0 No repository field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! serialport@4.0.1 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@4.0.1 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/username/.npm/_logs/2018-09-25T01_59_16_711Z-debug.log
username@nuc0A:~/Documents/arduinoPrograms$

reconbot, 2018-09-25:
Node SerialPort 4 is very old and doesn't support node 10.

jcabrerazuniga, 2018-09-25:
Can you please tell me the best node version compatible with Node SerialPort 4? Thanks

reconbot, 2018-09-25:
Node 4 I think, @noopkat can you upgrade firmata party? I thought you were at least on SerialPort 5

HipsterBrown, 2018-09-25:
@reconbot I can try opening a PR to `avrgirl-arduino` to upgrade the version of serialport being used.

HipsterBrown, 2018-09-25:
There is actually a work in progress PR to update serialport for `avrgirl-arduino`. https://github.com/noopkat/avrgirl-arduino/pull/160

noopkat, 2018-09-25:
thanks @HipsterBrown for sharing the PR 😄 

This has been a hairy upgrade task so far avrgirl-arduino, which is what powers firmata-party underneath. All three protocol packages have had incompatibilities with the newer SerialPort changes since v4 such as deprecated properties etc. Which is totally okay! But it's been a lot of extensive hardware testing that software driven unit tests cannot catch.

We're getting close, there's just a weird-not-reproducible-every-time bug on the Arduino Uno that I'm trying to get my head around (communication timeouts from the chip) and we need to ensure that Chrome app usage is also 💯 for Arduino to continue to use this package safely. This is also a little hard to test.

Anyway that's a little background. Would love to have Uno owners continue to test on a variety of different OS platforms 😉 I think we're really close.

reconbot, 2018-09-26:
Got a link for where we can help?

noopkat, 2018-09-30:
@reconbot I have now updated avrgirl-arduino to consume serialport v6 (v7 support coming soon!).

Thanks for flagging this issue @jcabrerazuniga, and thanks @HipsterBrown and @reconbot for bringing it to my attention.

reconbot, 2018-10-01:
Awesome! @jcabrerazuniga let us know if this works for you. I'll close for now.

closed 2018-10-01 by reconbot

jcabrerazuniga, 2018-10-02:
I was able to successfully run:

       npm install firmata-party -save

 and executed:

var five = require("johnny-five");
var board = new five.Board({
   port: "/dev/ttyACM0"
});

board.on("ready", function() {
  var led = new five.Led(13);
  led.blink(500);
});

board.on("error", function(err) {
   console.log(err);
})

while it seems the program is running, I got the next error message:node five1.js
1538473544568 Connected /dev/ttyACM0  
1538473554571 Device or Firmware Error A timeout occurred while connecting to the Board. 

Please check that you've properly flashed the board with the correct firmware.
See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting

If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seconds for complete reset, then run your program again.  
{ type: 'error',
  timestamp: 1538473554571,
  class: 'Device or Firmware Error',
  message:
   'A timeout occurred while connecting to the Board. \n\nPlease check that you\'ve properly flashed the board with the correct firmware.\nSee: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting\n\nIf connecting to a Leonardo or Leonardo clone, press the \'Reset\' button on the board, wait approximately 11 seconds for complete reset, then run your program again.',
  data: null }
Error: A timeout occurred while connecting to the Board.
    at Board.<anonymous> (/home/joseccz/Documents/arduinoPrograms/node_modules/johnny-five/lib/board.js:405:28)
    at ontimeout (timers.js:425:11)
    at tryOnTimeout (timers.js:289:5)
    at listOnTimeout (timers.js:252:5)
    at Timer.processTimers (timers.js:212:10)

Is there any special firmware I need to upload? how? Thx

reconbot, 2018-10-02:
You need to run firmata party instead of saving it of you haven't.

Try 

```
npx firmata-party
```

To run it and give it your board name. The default firmware is fine.

If it flashes and you still can't get j5 to work, please open an issue there as there are more people who can help debug.

jcabrerazuniga, 2018-10-03:
Working perfectly just as you indicated!!
Thanks a lot!!

reconbot, 2018-10-03:
Glad we could all help!