serialport/node-serialport #1686: On 'data' event is duplicating data across ports.

ChrisJVoss, 2018-10-01:
### Summary of Problem

Hi, I am writing an Electron/React App that interfaces with several scale devices. In the event that we may choose to add additional scales in the future, I am looping over the value returned by SerialPort.list(). I am then saving the appropriate port objects to my application's state and adding a serialport object with the matching path. Then, when the user presses 'start' the application loops over the saved state object and and calls the 'pipe', and 'on' methods for each serialport. The returned data values for each open serialport should then be saved to state and displayed.

Currently, when data is transmitted from one path, it is also showing up on the other path. In other words, scales 1 and 2 are on COM1 and COM2. Scale 1 has 10kg placed on it and the data from `.on('data', data => console.log(data))` for both COM1 and COM2 reads 10kg.

COM1 should show 10kg and COM2 should show 0kg.

### Code to Reproduce the Issue
```
class Test extends Component {
  constructor(props) {
    super(props);
    this.state = {
      parser: new Readline({ delimiter: '\r\n' }),
      scalePorts: {}
    };
    this.zeroOut = this.zeroOut.bind(this);
    this.startClick = this.startClick.bind(this);
  }
  componentDidMount() {
    this.helper();
  }

  async helper() {
    const allPorts = await SerialPort.list();
    let filteredPorts = {};
    allPorts.forEach(port => {
      if (port.manufacturer === 'FTDI') {
        filteredPorts[port.serialNumber] = Object.assign(
          {
            portPath: new SerialPort(port.comName, { lock: false })
          },
          port
        );
      }
    });
    this.setState({ scalePorts: filteredPorts });
  }

  startClick() {
    for (let port in this.state.scalePorts) {
      let path = this.state.scalePorts[port].portPath;
      path.pipe(this.state.parser).on('data', data => {
        console.log('2', port, data);
        let intermediary = Object.assign({}, this.state.scalePorts);
        intermediary[port].scaleData = data.slice(8, 13);
        console.log('3', intermediary[port].scaleData);
        this.setState({ scalePorts: intermediary });
      });
    }
  }
zeroOut() {
    console.log('clicked!');
    //port.write(Buffer.from([0x5a, 0xd, 0xa]));
    console.log(this.state.scalePorts);
  }

  render() {
    const scales = this.state.scalePorts;
    return (
      <div>
        <h1>Hello Scales!</h1>
        <h3>{scales.A9076Y4J ? scales.A9076Y4J.scaleData : 0.0}</h3>
        <button onClick={this.zeroOut}>Tare/Zero</button>
        <button onClick={this.startClick}>Start</button>
        <h3>{scales.AH05I0OT ? scales.AH05I0OT.scaleData : 0.0}</h3>
      </div>
    );
  }
}
```

### Versions, Operating System and Hardware
serialport v7.0.2
node v8.4.0
Mac v10.13.6
FTDI chipsets for rs232 to usb

Here is the output of the console logs in the `startClick()` function.  Only the scale on path 'A9076Y4J'  had any weight placed on it. Path 'AH05I0OT' had no device connected.
[image]

My gut tells me that this is because of the for...in loop but I am stumped as to how I could open an unknown number of paths and read their data.

ChrisJVoss, 2018-10-01:
This problem does not exist if I manually call `.on('data'...)` for each individual port. Only when I try to loop over the paths and call the `.on()` method automatically.

reconbot, 2018-10-01:
Maybe you want [`.once()`](https://nodejs.org/api/events.html#events_emitter_once_eventname_listener)?

ChrisJVoss, 2018-10-01:
@HipsterBrown 

Thank you for the suggestion; I gave it a shot but got the same results.
Here is how I call `.on('data' ...)` individually.

```
import SerialPort from 'serialport';
import Readline from '@serialport/parser-readline';

const port1 = new SerialPort('/dev/tty.usbserial-A9076Y4J');
const port2 = new SerialPort('/dev/tty.usbserial-AH05I0OT');
const parser1 = port1.pipe(new Readline({ delimiter: '\r\n' }));
const parser2 = port2.pipe(new Readline({ delimiter: '\r\n' }));
...
componentDidMount() {
    parser1.on('data', data => {
      let data1 = data.slice(8, 13);
      this.setState({ scale0: data1 });
      console.log('1', data1);
    });
    parser2.on('data', data => {
      let data2 = data.slice(8, 13);
      this.setState({ scale5: data2 });
      console.log('2', data2);
    });
  }
```
This way `data1` and `data2` are totally separate and I can log the individual results from each scale.
I could use the above approach if I knew the number of scales would never change but we may want to expand the system and I would like for it to be more flexible.

ChrisJVoss, 2018-10-01:
@reconbot 
So I substituted `.on()` for `.once()`, however, since my scales send a continuous stream of data, the listener is removed after the first data set is received.

HipsterBrown, 2018-10-01:
@ChrisJVoss In your individual example, you're creating a new Readline parser for each port, but in your original example you share `this.state.parser` between the ports. [The parser keeps an internal buffer](https://github.com/node-serialport/node-serialport/blob/master/packages/parser-delimiter/delimiter.js#L28), which leads me to believe the data could be getting mixed up there. Could you try creating a new parser for each port in the `for...in` loop?

@reconbot Does that sound reasonable?

ChrisJVoss, 2018-10-01:
@HipsterBrown 
That was it!! Thank you so much!

closed 2018-10-01 by ChrisJVoss

reconbot, 2018-10-01:
Better than reasonable it worked!