serialport/node-serialport #1687: Port doesn't seem to open?

jdshin123, 2018-10-01:
### Summary of Problem

(Please answer all 3)

- What are you trying to do?
 * write and read to/from a comport.
- What happens?
 * Does not seem to be able to open a port. When i write to comport, the message stays in the buffer but
never gets sent. When i connect to comport via terminal im able to see the message that has not been sent. I also tried closing the port i.e port.close() but i get an error saying port is not open.
- What should have happened? 
* The message should have been able to go through and receive message

### Code to Reproduce the Issue

[image]

const serialPort = require('serialport');
const readLine = require('@serialPort/parser-readLine');

serialPort.list(function(err,ports){
  ports.forEach(function(port){
    console.log(port.comName);
  })
})

var port = new serialPort('com5',{
  autoOpen: true,
  baudRate: 9600,
  dataBits: 8,
  partiy: 'none',
  stopBits: 1
})

const parser = port.pipe(new readLine());

 parser.on('data',console.log)

 port.write('fakemessage',function(err){
   
   if(err){
     console.trace(err);
   }
   console.log('message written..')
 });

 port.open(function(err){
  if(err){
    console.log(err);
  }
});
 port.drain();

 port.close(function(err){
   if(err){
     console.log(err)
   }
 });

### Versions, Operating System and Hardware

- SerialPort@? 7.0.2
- Node.js v? 9.40
- Windows? Linux? Mac? windows 10 (home)

HipsterBrown, 2018-10-02:
@jdshin123 Can you confirm that 'COM5' is listed when you run `Serialport.list()`?

Also, unless you pass the `autoOpen: false` option to the `Serialport` construction, then the following code is redundant and will probably throw an error:

```es6
var port = new serialPort('com5',{
  autoOpen: true,
  baudRate: 9600,
  dataBits: 8,
  partiy: 'none',
  stopBits: 1
})
port.open(function(err){
  if(err){
    console.log(err);
  }
});
```

jdshin123, 2018-10-02:
@reconbot @HipsterBrown  welp.. i'm such a dolt.. you were right! it was a "\n" that was missing.. When i opened the terminal , i had to press enter for the message to display.. Thank you guys!

closed 2018-10-02 by jdshin123