serialport/node-serialport #1688: Check SerialPort onData ends to write

jnsanchezg, 2018-10-02:
### Summary of Problem

Hello, im using serialport, and it's amazing. But i have a little problem, because actually I get a data from a port, but i don't know when onData function ends to write. I'm using Express to create a new instance of serialport and get the returned value from onData function and this is the code.

This is my code:

```js

var SerialPort = require('serialport');
var express = require('express');
var NodeCache = require("node-cache");
var fs = require('fs');

var app = express();

//Variables Globales
var _serialPort = null;
var _dataSP = "";
var _dataSP2 = "";
var _isError = false;
var _errorMessage = "";

//Creates a new serialport instance 
app.get('/open_serial_connection', function (req, res) {
    var lMessage = "";
    var lError = 0;
    try {
        if (_serialPort == null) {
            _serialPort = new SerialPort("COM1", {                
                baudRate: 9600,
                //parity:    , 
                //dataBits:  ,
                //stopBits:  ,
                autoOpen: false
            });
            _serialPort.open();
            _serialPort.on("error", onError);
            _serialPort.on("open", onOpen);
            _serialPort.on("data", onData);
            _serialPort.on("close", onClose);
            lMensaje = "Success.";
        } else {
            //Checks if serialport is actually opened           
            if (_serialPort.isOpen) {

                lError = 1;
                lMenssage = "Connection is open.";
            } else {
                lMessage = "No connections available.";
            }
        }

    } catch (e) {
        lMessage = "Error trying to create a serialport instance";
        lError = 1;
    }
    res.send({ Error: lError, Message: lMessage});
});

function onOpen(data) {
...
}

function onClose(err) {
...
}
function onData(data) {
    //console.log(data.lenght);
    _datoSP = data.toString('utf-8');
    _datoSP = _datoSP.replace(/(\r\n\t|\n|\r\t)/gm, " ");
    _datoSP2 += _datoSP.replace(/(\r?\n|\r)/gm, '') + " ";
    //console.log("length de SP2: "+_datoSP2.length);
    console.log(_datoSP2);
}

function onError(data) {
    _errorMessage = data.toString();
    _isError = true;
    console.log(data);
}

app.get('/result_connection_serial', function (req, res) {
    res.send({ Result : _dataSP2 });
});

```
With this code, I can get the data from the serialport connection but my actual problem is check when all the data returned from any serialport equip ends to send the data. 

Any idea? 
Thanks.

### Versions, Operating System and Hardware
- SerialPort@6.2.2
- Node.js v8.12.0
- Windows

HipsterBrown, 2018-10-02:
@jnsanchezg Because `Serialport` is based on a Node.js duplex steam, ([docs reference](https://serialport.io/docs/api-stream)), there _should_ be an `end` event is ["emitted when there is no more data to be consumed from the stream"](https://nodejs.org/api/stream.html#stream_event_end). It is probably not documented by Serialport because Node.js already documents it in their [stream docs](https://nodejs.org/api/stream.html). /cc: @reconbot for confirmation

jnsanchezg, 2018-10-02:
@HipsterBrown ,thanks for the answer, so adding something like:
```js
app.get('/open_serial_connection', function (req, res) {
...
_serialPort.on("end", onEnd);
...
});
....
function onEnd(){
  console.log("Ends")
}
```
This code should work?  Actually I'm making tests using a Codebar Lector but when i get the data, it doesn't display the console.log from onEnd.

reconbot, 2018-10-02:
End might happen when you close the port. It doesn't happen because time
has passed after the the last data from the port.

jnsanchezg, 2018-10-02:
@reconbot this is my result in console:

**
$ node index.js
App in port 3000.

Open:
75.5455
75.5455 75.5455
75.5455 75.5455 100
75.5455 75.5455 100 13.25
75.5455 75.5455 100 13.25 13.25
Closed**

HipsterBrown, 2018-10-02:
@jnsanchezg what type of device are you reading data from? Is there any sort of ending delimiter? See: https://serialport.io/docs/api-parser-delimiter

jnsanchezg, 2018-10-02:
@HipsterBrown I handle with 3 types of devices:
A bar reader.
A digital refractometer.(This one returns something like **001 Result = 45.40**, if i get this data, first it gets **001 Result**, and then **= 45.40**, so I'm concatenate the result in a global variable)
A digital weighing scale(This one returns always a value, it's continuous, and I use the same idea in the digital refractometer).

HipsterBrown, 2018-10-02:
You might be able to use the Readline parser to get the complete read `.on('data')`. Then you don't need to do the `String.replace` manually in your `onData` callback. 

https://serialport.io/docs/api-parser-readline

jnsanchezg, 2018-10-03:
Got it, I'm going to use in those devices **Readline** and them I'll share the results.

jnsanchezg, 2018-10-03:
@HipsterBrown using Readline works very well in the three devices, but I forgot to mention another device(a **_digital titrator_**).
This one throws this data:
$S PC/LIMS V1\r\n$S Devices V1\r\n$S device 848 Titrino plus V1\r\nP 5.848.0025\tS 32650\r\n$S ADC1\r\nADC 3.848.1210\tS 94849\r\n$E\r\n$S MSB1\r\n$S 848\r\nS 85896\tS 30017659\r\n$E\r\n$S 801\r\nS 82544\t\r\n$E\r\n$S 848\r\n\t\r\n$E\r\n$E\r\n$E\r\n$E\r\n$S Sample data V1\r\n\t\t1.0\tg\r\n$E\r\n$S MPL V2\r\n$S Mode 1\t04\tSET pH\tV1.0\r\n1\t0.0\t4.019\t0.0000\t0.0\t25.0\r\n2\t2.0\t4.057\t0.1220\t3108.8\t25.0\r\n3\t4.0\t4.065\t0.1580\t875.3\t25.0\r\n4\t6.0\t4.085\t0.2160\t1494.6\t25.0\r\n5\t8.0\t4.118\t0.2960\t2428.6\t25.0\r\n6\t10.0\t4.158\t0.3660\t2172.2\t25.0\r\n7\t12.0\t4.189\t0.4400\t2103.8\t25.0\r\n8\t14.0\t4.225\t0.5120\t2334.6\t25.0\r\n9\t16.0\t4.264\t0.5840\t2042.2\t25.0\r\n10\t18.0\t4.300\t0.6560\t2144.6\t25.0\r\n11\t20.0\t4.336\t0.7280\t2156.8\t25.0\r\n12\t22.0\t4.371\t0.8020\t2171.6\t25.0\r\n13\t24.0\t4.408\t0.8760\t2314.8\t25.0\r\n14\t26.0\t4.438\t0.9420\t1958.4\t25.0\r\n15\t28.0\t4.474\t1.0240\t2418.9\t25.0\r\n16\t30.0\t4.510\t1.0920\t2120.0\t25.0\r\n17\t32.0\t4.544\t1.1600\t2091.2\t25.0\r\n18\t34.0\t4.583\t1.2320\t2112.3\t25.0\r\n19\t36.0\t4.615\t1.2980\t1950.3\t25.0\r\n20\t38.0\t4.656\t1.3800\t2463.6\t25.0\r\n21\t40.0\t4.684\t1.4360\t1782.1\t25.0\r\n22\t42.0\t4.719\t1.5140\t2192.2\t25.0\r\n23\t44.0\t4.765\t1.5780\t2131.9\t25.0\r\n24\t46.0\t4.792\t1.6380\t1643.2\t25.0\r\n25\t48.0\t4.838\t1.7180\t2468.4\t25.0\r\n26\t50.0\t4.870\t1.7700\t1609.0\t25.0\r\n27\t52.0\t4.904\t1.8440\t2107.8\t25.0\r\n28\t54.0\t4.952\t1.9000\t1896.6\t25.0\r\n29\t56.0\t4.983\t1.9620\t1636.5\t25.0\r\n30\t57.9\t5.035\t2.0360\t2361.4\t25.0\r\n31\t60.0\t5.064\t2.0800\t1397.3\t25.0\r\n32\t61.9\t5.102\t2.1460\t1776.3\t25.0\r\n33\t63.9\t5.150\t2.2040\t1896.1\t25.0\r\n34\t65.9\t5.186\t2.2600\t1559.4\t25.0\r\n35\t67.9\t5.236\t2.3160\t1831.1\t25.0\r\n36\t69.9\t5.280\t2.3720\t1550.6\t25.0\r\n37\t71.9\t5.316\t2.4260\t1701.9\t25.0\r\n38\t73.9\t5.360\t2.4860\t1714.5\t25.0\r\n39\t75.9\t5.406\t2.5380\t1660.1\t25.0\r\n40\t77.9\t5.448\t2.5940\t1596.1\t25.0\r\n41\t79.9\t5.498\t2.6440\t1581.4\t25.0\r\n42\t81.9\t5.538\t2.6960\t1494.5\t25.0\r\n43\t83.9\t5.592\t2.7500\t1694.5\t25.0\r\n44\t85.9\t5.632\t2.7940\t1326.4\t25.0\r\n45\t87.9\t5.681\t2.8500\t1635.0\t25.0\r\n46\t89.9\t5.724\t2.8940\t1447.0\t25.0\r\n47\t91.9\t5.772\t2.9500\t1519.9\t25.0\r\n48\t93.9\t5.816\t2.9960\t1518.9\t25.0\r\n49\t95.9\t5.871\t3.0480\t1504.2\t25.0\r\n50\t97.9\t5.908\t3.0960\t1385.4\t25.0\r\n51\t99.9\t5.978\t3.1460\t1608.9\t25.0\r\n52\t101.9\t6.018\t3.1820\t1045.8\t25.0\r\n53\t103.9\t6.062\t3.2380\t1603.0\t25.0\r\n54\t105.9\t6.118\t3.2820\t1448.8\t25.0\r\n55\t107.9\t6.172\t3.3240\t1212.2\t25.0\r\n56\t109.9\t6.221\t3.3720\t1363.2\t25.0\r\n57\t111.9\t6.270\t3.4160\t1373.3\t25.0\r\n58\t113.9\t6.332\t3.4580\t1278.4\t25.0\r\n59\t115.9\t6.380\t3.4960\t1118.7\t25.0\r\n60\t117.9\t6.440\t3.5440\t1437.3\t25.0\r\n61\t119.9\t6.500\t3.5760\t1121.1\t25.0\r\n62\t121.9\t6.557\t3.6180\t1102.8\t25.0\r\n63\t123.9\t6.629\t3.6500\t1163.5\t25.0\r\n64\t125.9\t6.695\t3.6860\t942.8\t25.0\r\n65\t127.9\t6.758\t3.7220\t1119.0\t25.0\r\n66\t129.9\t6.834\t3.7520\t984.5\t25.0\r\n67\t131.9\t6.899\t3.7860\t927.0\t25.0\r\n68\t133.9\t6.984\t3.8140\t1005.6\t25.0\r\n69\t135.9\t7.040\t3.8420\t776.0\t25.0\r\n70\t137.9\t7.129\t3.8760\t984.0\t25.0\r\n71\t139.9\t7.208\t3.8980\t882.9\t25.0\r\n72\t141.9\t7.280\t3.9280\t741.6\t25.0\r\n73\t143.9\t7.390\t3.9560\t912.6\t25.0\r\n74\t145.9\t7.445\t3.9760\t740.2\t25.0\r\n75\t147.9\t7.511\t4.0100\t785.3\t25.0\r\n76\t149.9\t7.633\t4.0340\t1018.2\t25.0\r\n77\t151.9\t7.706\t4.0540\t699.0\t25.0\r\n78\t153.9\t7.783\t4.0840\t691.0\t25.0\r\n79\t155.9\t7.863\t4.1080\t841.1\t25.0\r\n80\t157.9\t7.944\t4.1380\t781.7\t25.0\r\n81\t159.9\t8.027\t4.1620\t862.2\t25.0\r\n82\t161.9\t8.097\t4.1740\t774.4\t25.0\r\n83\t163.9\t8.104\t4.1740\t690.9\t25.0\r\n84\t165.9\t8.105\t4.1740\t623.6\t25.0\r\n85\t167.9\t8.104\t4.1740\t559.5\t25.0\r\n86\t169.8\t8.101\t4.1740\t267.6\t25.0\r\n87\t171.9\t8.130\t4.1940\t146.0\t25.0\r\n88\t173.9\t8.143\t4.1940\t140.2\t25.0\r\n89\t175.8\t8.145\t4.1940\t134.9\t25.0\r\n90\t177.8\t8.145\t4.1940\t130.0\t25.0\r\n91\t179.8\t8.144\t4.1940\t143.9\t25.0\r\n92\t181.8\t8.143\t4.1940\t19.1\t25.0\r\n93\t182.8\t8.143\t4.1940\t17.5\t25.0\r\n$E\r\n$E\r\n$S DETERM V1\r\n$S Props V2\r\nAcidez\tsaved\t20181003-103746\t326501538563066000\t2018-10-03 10:37:46\toriginal\tRegular without errors\tTicofrut\t61\t\t\t\t\t\t\t\t\t\t\t\t\t\t\t\t\t\t\t\t\t\t\t\t\t\t\t\t\r\n$E\r\n$S Results V1\r\n$S Result 1C.R1 V1\r\nAcidez mL\t4.194\tmL\t0\r\n$E\r\n$S Result 1C.R2 V1\r\n\t\tmeq/L\t-1\r\n$E\r\n$S Result 1C.R3 V1\r\n\t\tg/L\t-1\r\n$E\r\n$S Result 1C.R4 V1\r\n\t\t%\t-1\r\n$E\r\n$S Result 1C.R5 V1\r\n\t\t%\t-1\r\n$E\r\n$E\r\n$S Mode 1\t04\tSET pH\tV2.0\r\n$S EP V1\r\n4.1940\t8.143\t\t182.8\t25.0\t1\r\n$E\r\n$S Cal V1\r\n$E\r\n$S StdAdd V1\r\n$E\r\n$E\r\n$S Other Variables V1\r\n$S System V1\r\n195.0\t61\t1426\t0\t\t\t0\t\t0\r\n$E\r\n$S Mode 1\t04\tSET pH\tV2.0\r\n1.000\t1.000\t4.1940\t183.9\t0.0000\t4.024\t25.0\t4.019\t25.0\t1.1\t8.143\t25.0\t\t\t\t6.963\t99.2\t\t93\t\t\tRegular stop\tmanual\t\t\tNaOH.\r\n$E\r\n$S Statistic V2\r\n$E\r\n$S ComVar local V1\r\n$E\r\n$S Titrant V1\r\n$S Titrant1 V1\r\nNaOH.\t1.000\tmol/L\t1.000\t\t\t2018-06-11 15:56:11\t30017659\t20\t495484\r\n$E\r\n$E\r\n$S Sensor V1\r\n$S Sensor1 V2\r\npH electrode\tpH\t\t\t99.2\t6.963\t25.0\t(manual)\t2018-10-03 07:00:45\tCAL pH\tTicofrut\r\n$E\r\n$E\r\n$E\r\n$S [... cut ...] ssage\t\t\r\n$E\r\n$S R4\r\n4\t\t\t2\t%\t3\toff\toff\t3\ton\toff\tRound\toff\tinvalid\tinvalid\tDisplay message\t\t\r\n$E\r\n$S R5\r\n5\t\t\t2\t%\t4\toff\toff\t4\ton\toff\tRound\toff\tinvalid\tinvalid\tDisplay message\t\t\r\n$E\r\n$E\r\n$S 07\tT+ Report\t\r\noff\toff\toff\toff\toff\r\n$E\r\n$E\r\n$E\r\n

From this result, I can't use Readline, because it throws the result in parts. 
If I use a [(https://serialport.io/docs/api-parser-regex)] regex to get the last one \r\n, it should work?
Thanks!

HipsterBrown, 2018-10-03:
Good to hear the Readline parser works for those three devices. 

I think the Regex parser is worth a shot for that other device given the odd output. :+1:

Can we close out this issue?

jnsanchezg, 2018-10-03:
@HipsterBrown but now I have another question? Using a **regex**, this one applies under a specific buffer? It is because all the previous data is a lot of Buffers.
Thanks.

jnsanchezg, 2018-10-03:
In wich way?

closed 2018-12-10 by stale[bot]