serialport/node-serialport #1689: Buffer/baud rate limitations?

TmpR, 2018-10-02:
I'm trying to write an RGB LED matrix controller to run in Node v8.12 on a RasPi using UART on the GPIO pins to talk to an Arduino and I keep experiencing "strangeness".

To keep things simple, I'm using the [Glediator](http://www.solderlab.de/index.php/software/glediator) protocol. I can run Glediator on the RasPi and everything works fine.  It's set to 1000000 baud and it's currently controlling 253 RGB LEDs. With the addition of the "start of frame" byte, this is 760 bytes.  In future, I hope to control 1,000 LEDs.

Gladiator runs in Java and works perfectly, however, I can't get anything in Node to work. As soon as I start sending multiple frames, the data seems to randomly corrupt. If I reduce the number of LEDs to around 50, it appears to run smoothly.  And when it does corrupt the corruption only seems to occur after the first 50 LEDs' worth of data.  I've tried sending just 26 LEDs at a time (1 row) and flushing the buffer but it still corrupts at the same point?

I'm using the [raspi-serial](https://www.npmjs.com/package/raspi-serial) wrapper package for simplicity.

I've tried lots of different ways of sending the data using node but I'm wondering if anyone could help with some experience.  The hardware and OS are not the issue as the Java based Glediator works fine. So it must be my code, this package or the Node stack that's at fault.

Does anyone know of any buffer,baud,bandwidth or latency limitations with Node or this package?

Can anyone suggest how to stream a 760 byte buffer, repeatedly, as fast as possible but executing some user-code to modify the buffer between each transmission. 

Thanks!

HipsterBrown, 2018-10-02:
@TmpR Can you post information about the OS on the Pi and Node.js version you're using? Some example code can also help with debugging. 

Is this the matrix controller you're using? http://www.solderlab.de/index.php/hardware/matrix-controller-board

There is a relevant note in the `node-pixel` package about running multiple strips -> https://www.npmjs.com/package/node-pixel#a-note-on-multiple-strips

TmpR, 2018-10-02:
The OS is the latest Rasbian installed via NOOBS. Node v8.12.

The controller is an Arduino Pro Mini, but will be replacing with a Mega (more memory). The firmware on it is about 3 lines of code, that reads the byte stream and uses FastLED to control the WS2812b/NeoPixel LEDs.  This works with the Glediator software.

There are limits with the number of pixels, mainly memory on the Arduino, but according to Glediator you can run up to 1024 pixels on an Arduino Mega. They recommend 1000000 baud for 1024 pixels, so I'm starting with this baud so it's one less thing to change when I scale up from 253 to 1012 pixels.

1012 pixels @ 25fps = about 600kbps so about 60% utilisation of 1000000 baud if my understanding is correct.

HipsterBrown, 2018-10-02:
> The controller is an Arduino Pro Mini, but will be replacing with a Mega (more memory)

Then you could probably use `node-pixel` via the backpack setup -> https://github.com/ajfisher/node-pixel/blob/HEAD/docs/installation.md#i2c-backpack

reconbot, 2018-10-02:
You'll hit a CPU limit with nodejs and that will speak to available bandwidth. It's not impossible you're hitting that, but you should test and let us know. Performance hasn't been a goal of this package in the past but it's not not a goal.

raspi-serial uses serialport ^6.2.0, [6.2.1 fixed a memory leak](https://github.com/node-serialport/node-serialport/blob/master/packages/serialport/CHANGELOG.md#621-2018-06-28)

Get some stats and we can help debug. Code also helps.

closed 2018-12-09 by stale[bot]