serialport/node-serialport #1690: Consider "end" even in addition to "close" on @serialport/stream

reconbot, 2018-10-02:
## 💥 Proposal

From https://github.com/node-serialport/node-serialport/issues/1688#issuecomment-426316689 it might be worth adding the `end` event to the stream. We would do it to work well with other streams. @HipsterBrown do you think it's needed?

HipsterBrown, 2018-10-02:
If we're keeping the Stream interface as an implementation of the Duplex stream, then `end` could probably be emitted once zero bytes are read -> https://github.com/node-serialport/node-serialport/blob/master/packages/stream/stream.js#L380

Helps keep things consistent with how people learn streams, i.e. https://medium.freecodecamp.org/node-js-streams-everything-you-need-to-know-c9141306be93#a9a7

reconbot, 2018-10-02:
The bindings literally never return 0 bytes, it's the spec that they have to return at least a single byte or wait for data. The only time the stream ends is on close.

reconbot, 2018-10-02:
And once a stream has ended I'm pretty sure it can't be started again.

HipsterBrown, 2018-10-02:
So is the check for 0 bytes in the stream source code just for safe measure?

```es6
this.binding.read(pool, start, toRead).then(
    bytesRead => {
      debug('binding.read', `finished`)
      // zero bytes means read means we've hit EOF? Maybe this should be an error
      if (bytesRead === 0) {
        debug('binding.read', 'Zero bytes read closing readable stream')
        this.push(null)
        return
      }
      pool.used += bytesRead
      this.push(pool.slice(start, start + bytesRead))
    },
    err => {
      debug('binding.read', `error`, err)
      if (!err.canceled) {
        this._disconnected(err)
      }
      this._read(bytesToRead) // prime to read more once we're reconnected
    }
  )
```

reconbot, 2018-10-02:
I guess so, I don't really remember =)

It's possible that we have the read return 0 bytes if it's pending and the
port is closed. Because it's not an error in that case.
---
Francis Gulotta
wizard@roborooter.com

reconbot, 2019-08-30:
Worked in #1926 thinking it will be an option to enable or disable this behavior. If we have it on by default we need a HUGE warning in the changelog.

rotu, 2023-05-08:
It seems unexpected to have this off by default. It means that, while `SerialPortStream` seems to implement `stream.Readable`, it does not obey the [expected behavior](https://nodejs.org/api/stream.html#event-end).

Similarly, the [`finish` event](https://nodejs.org/api/stream.html#event-finish) does not appear to be implemented.