serialport/node-serialport #1691: Make `npm install serialport` just work on Windows

Masterxilo, 2018-10-02:
## 💥 Proposal

`npm install serialport` should just work on Windows, no compiler required.

## Motivation
`npm install serialport` should just work on any computer and commandline where some reasonably recent version of `node` and `npm` are installed and in the PATH.

## Pitch

Both `npm install serialport` and `npm install serialport-win-fix` failed for me in `cmd.exe`. (They work fine in `bash` from the WSL).

I just want to talk to a COM device, this should not be hard in any language or platform...

Masterxilo, 2018-10-02:
Error logs
```
npm WARN enoent ENOENT: no such file or directory, open 'C:\Users\pau\package.json'
npm WARN pau No description
npm WARN pau No repository field.
npm WARN pau No README data
npm WARN pau No license field.

npm ERR! path C:\Users\pau\node_modules\.bin\serialport-list
npm ERR! code UNKNOWN
npm ERR! errno -4094
npm ERR! syscall lstat
npm ERR! UNKNOWN: unknown error, lstat 'C:\Users\pau\node_modules\.bin\serialport-list'

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\pau\AppData\Roaming\npm-cache\_logs\2018-10-02T16_29_44_326Z-debug.log
```

```
npm WARN deprecated hoek@2.16.3: The major version is no longer supported. Please update to 4.x or newer

> serialport-win-fix@4.0.16 install C:\Users\pau\node_modules\serialport-win-fix
> node-pre-gyp install --fallback-to-build

Der Befehl "node-pre-gyp" ist entweder falsch geschrieben oder
konnte nicht gefunden werden.
npm WARN enoent ENOENT: no such file or directory, open 'C:\Users\pau\package.json'
npm WARN pau No description
npm WARN pau No repository field.
npm WARN pau No README data
npm WARN pau No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! serialport-win-fix@4.0.16 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport-win-fix@4.0.16 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\pau\AppData\Roaming\npm-cache\_logs\2018-10-02T16_31_25_414Z-debug.log
```

It would be hugely appreciated if you guys could fix this stuff once for all the world. I have used other node modules with native components that just worked seamlessly in the past, it would be great if this could become one of them.

Masterxilo, 2018-10-02:
```
npm --version && node --version
6.2.0
v10.9.0
```

HipsterBrown, 2018-10-02:
@Masterxilo Are you trying to use the command line tools or install `serialport` as a dependency for a project?

Command line tools: https://serialport.io/docs/guide-cli

The native modules are precompiled for windows (https://github.com/node-serialport/node-serialport/releases/tag/%40serialport%2Fbindings%402.0.2) and you can see a list of support environments here -> https://serialport.io/docs/guide-platform-support

Also, the `serialport-win-fix` package is hundreds of commits behind `serialport`, so we cannot help with any issues around that fork.

closed 2018-12-09 by stale[bot]