serialport/node-serialport #1692: serialport not "fully" opening a connection with Arduino Micro

SterlingButters, 2018-10-04:
### Summary of Problem
I am trying to use my Arduino Micro as an HID device for keyboard input using serialport. If I simply open the serial monitor from the Arduino IDE and then run my code `node keyboard-transmit.js` then everything works perfectly and the letter `A` is written by the cursor. If I do this without first opening the serial monitor in the IDE then I get nothing written by the cursor however if I were to open the monitor after running the code, the serial output is shown there in the monitor.

My question is 1) what is happening to the connection by opening the serial monitor in the IDE and 2) can I do it in serialport so that I do not have to manually open the serial monitor from the IDE?

### Code to Reproduce the Issue
Javascript:
```
var SerialPort = require('serialport');
var port = new SerialPort('/dev/cu.SLAB_USBtoUART', {
  baudRate: 9600,
  autoOpen: false,
});

port.open(function (err) {
  if (err) {
    return console.log('Error opening port: ', err.message);
  }
});

port.on('open', function() {
  // Keyboard
  port.write(Buffer.from('A'));
});
```

Arduino:
```
#include <SoftwareSerial.h>
#include <Keyboard.h>

SoftwareSerial rpiSerial(10, 11); // RX, TX

void setup() {
  // Open serial communications and wait for port to open:
  Serial.begin(9600);
  // Set the data rate for the SoftwareSerial port
  rpiSerial.begin(9600);
  // initialize control over the keyboard:
  Keyboard.begin();
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  Serial.println("Serial Connection Established");
}

void loop() {
  if (rpiSerial.available()) {
    // read incoming serial data:
    char inChar = rpiSerial.read();
    // Type the ASCII value received:
    Keyboard.println(inChar);
    Serial.println(inChar);
  }
}
```

### Versions, Operating System and Hardware

- SerialPort@ 2.0.1
- Node.js v10.10.0
- macOS Mojave
- Arduino Micro ATmega32U4

HipsterBrown, 2018-10-04:
@SterlingButters `Serialport#write` takes a callback that is "called once the write operation finishes." As well as:

> Returns a boolean false if the stream wishes for the calling code to wait for the drain event to be emitted before continuing to write additional data

Checking the return value of your `port.write(Buffer.from('A'))` call and adding a callback with some logging, should give you more info about if the port is open when writing to the device. 

You can also check out the [debugging output](https://serialport.io/docs/guide-debugging) from serialport while your program is running. If you can share that here, it can help with figuring out the issue.

SterlingButters, 2018-10-04:
I know that the data is being written though, its just that the action `Keyboard.println(inChar);` is not taken in the arduino script but `Serial.println(inChar);` is. 

Debugging:

```
bash-3.2$ DEBUG=* node keyboard-transmit.js
  @serialport/bindings loading DarwinBinding +0ms
  @serialport/stream opening path: /dev/cu.SLAB_USBtoUART +0ms
  @serialport/binding-abstract open +0ms
  @serialport/bindings:poller Creating poller +0ms
  @serialport/stream opened path: /dev/cu.SLAB_USBtoUART +9ms
  @serialport/stream _write 1 bytes of data +1ms
  @serialport/binding-abstract write 1 bytes +9ms
  @serialport/bindings:unixWrite Starting write 1 bytes offset 0 bytesToWrite 1 +0ms
  @serialport/bindings:unixWrite write returned null 1 +1ms
  @serialport/bindings:unixWrite wrote 1 bytes +0ms
  @serialport/bindings:unixWrite Finished writing 1 bytes +0ms
  @serialport/stream binding.write write finished +1ms```

SterlingButters, 2018-10-07:
As a workaround (for now) I inserted the following into my code: 
```
var spawn = require("child_process").spawn;
var process = spawn('pio', ["device", "monitor", "--port", "/dev/cu.usbmodemHIDPC1"])
```

Though it is important to notice that the port I am actually sending data to is `/dev/cu.SLAB_USBtoUART` as can be seen below:

```
// TODO: Figure out why I need this
var spawn = require("child_process").spawn;
var process = spawn('pio', ["device", "monitor", "--port", "/dev/cu.usbmodemHIDPC1"])

// Set up Serial connection
var serialPort = require('serialport');
var port = new serialPort('/dev/cu.SLAB_USBtoUART', {
  baudRate: 9600,
  autoOpen: false,
});

// Report error if Serial connection cannot be made
port.open(function (err) {
  if (err) {
    return console.log("Error opening port: ", err.message);
  }
});

port.on('open', function() {
  console.log("Serial connection made")
  setInterval( function() {
  var msg = 'test';
  // console.log(msg.toString('utf8'))
  port.write(msg.toString('utf8'));
  }, 1000);
});
```

Lucky for me, platformio in atom text editor has a serial monitor which gave me command I spawned in the first two lines of code. It appears the command initiates a miniterm instance on the other port: 

```
--- Miniterm on /dev/cu.usbmodemHIDPC1  9600,8,N,1 ---
``` 

Maybe someone can explain why this is necessary and a cleaner way to move forward?

HipsterBrown, 2018-10-08:
@SterlingButters To clarify, you're currently expecting serialport to send a character to the Arduino Micro, which should then send that keystroke back to your computer and show in your terminal? 

I'm a little confused about the setup because I see some code related to a raspberry pi as well.

SterlingButters, 2018-10-08:
@HipsterBrown Since I dont have access to my RPi3 at the moment I am having to use my MBP as both the sender and receiver of the data so your understanding is spot on: the laptop is currently emulating the RPi3.

SterlingButters, 2018-10-08:
Are you thinking it might work from the RPi3 without the additional command?

SterlingButters, 2018-10-09:
Thanks for the tip but for my application it is necessary as constant changes will need to be handled often. `Keyboard.write` yields the same as `Keyboard.println`; although, once I changed it to `write` things stopped working as expected for a little bit but now everything works as before (Idk what happened but scared to change it again) - point being I still need to "initialize" the serial monitor from the CLI.

closed 2018-12-17 by stale[bot]