serialport/node-serialport #1697: v6.2.2 binary not working in Windows

partap, 2018-10-09:
### Summary of Problem

- What are you trying to do?

Using serialport as a dependency in my electron project, I'm developing on MacOS and building bundles for Linux and Windows.  I was using `^6.2.0` for serialport version.

- What happens?

I recently re-ran yarn to upgrade some dependencies, and serialport was bumped up to v6.2.2.  This seems to work fine on the Mac.  But when I built a Windows 64 bit package and tried to run it, it exits immediately and I see this error in the log:
```
error: uncaughtException: \\?\C:\Users\partap\AppData\Local\Temp\b4b31dd3-0edf-4688-a81b-9e877893c1b6.tmp.node 
is not a valid Win32 application\\?\C:\Users\partap\AppData\Local\Temp\b4b31dd3-0edf-4688-a81b-9e877893c1b6.tmp.node 
```

- What should have happened?

If I back off to v6.2.0 or v6.2.1, the windows executable runs normally.

### Code to Reproduce the Issue

As far as I can tell, it breaks at:
```
var serialport = require('serialport')
```
Initially I thought that the error was referring to the entire electron binary, but since I saw the error in the log, it seems that the program runs long enough to start up the logger, anyways.

### Versions, Operating System and Hardware

- SerialPort@6.2.2?
- Node.js v?8.12.0
- MacOS building for Windows with Electron@2.0.11 and electron-builder@20.28.4
- Hardware and chipset? Prolific serial USB adapter

HipsterBrown, 2018-10-09:
@partap So I think the issue is that a prebuilt binary for `serialport-v6.2.2-electron-v57-win32-x64` was never built, where it exists for the v6.2.1 and v6.2.0 releases. 

v6.2.2 Release: https://github.com/node-serialport/node-serialport/releases/tag/v6.2.2

I'm looking into ways to create that binary and add it to the release. If you have a Windows system handy, you can try to build from source: https://serialport.io/docs/guide-installation#windows

partap, 2018-10-10:
@HipsterBrown Thanks for the quick response!

Yeah, I tried building my app with v6.2.2 directly on Windows a little while after posting this issue and ran into a message complaining about needing vcbuild.exe or something. So I guessed that there's just not an appropriate binary available.

Last time I installed Visual Studio it wanted about 15GB(!)... This is just a little Windows VM I keep on my Mac for testing, so I'm not trying to double its image size.  I can get by with the older binary for now ;) . Not sure if there's anything I need to update the module for, anyways...

...Unless...I just realized today that the Windows build of my project is not properly detecting CTS or DSR pins with `serialport.get()`.  I'm using them to determine if certain devices are actually connected to the serial port so the program doesn't wait around for a response for a timeout (video projectors...can take over a minutes to respond sometimes if cold).

I was assuming that the problem was with the Windows driver for my USB-Serial adapter, but I got a report from a colleague of this happening on a Windows machine with a native serial port.  This is actually the reason I needed a new Windows build in the first place...to make the CTS/DSR detection optional.

They are actually using a different model of projector than I have access to, so it might just be that it just doesn't set CTS like the same-branded brethren I'm using here, but is it possible that this is an issue with the Windows build of `serialport`? (and that it's fixed in the one newer patch-level version that I can't use? lol)

I'll be able to do better testing soon once I get some actual Windows hardware hooked up.  If there's an actual problem with `serialport.get()` I'll open a separate issue.

closed 2018-12-17 by stale[bot]