serialport/node-serialport #1698: How to properly unit test with mock bindings

bdmayes, 2018-10-11:
### Summary of Problem
I'm having a bit of trouble trying to get a unit test of this going. Basically, I have an "adapter" that wraps a serial port (ultimately running on an embedded Linux device). Right now, my firmware team is still working on their end of things, but we have some agreed upon commands, so I am trying to mock out their device, and then test that when I invoke functions in my adapter, that under the covers it creates buffers correctly and passes them over to the device. This will allow me to have a unit test that we can run that will all be green. They can then code up their firmware and run my integration test target and be red. As they code up firmware and get it to respond in the agreed upon format, their tests should slowly turn from red to green -- and if there is a discrepancy we can look at what I thought was supposed to happen (in the unit test), and compare it to the response from the actual device (integration test).

I turned on the echo feature so that my serial port will echo back the input data. Maybe the `write` event is simply the wrong event to be listening to? I'm trying to dig in and see exactly what happens in that scenario, but I'll keep trying to look into it (hopefully while someone can answer this).

(Please answer all 3)

- I'm just trying to use the mock bindings to write some unit tests so that I can verify that a function call is creating the correct data buffers that ultimately will be passed into a serial device.
- Right now what happens is that I'm mocking out the serialport binding and my adapter invokes the `write` function. However, I keep trying to listen to an event to see when that happens and I'm not getting one.
- What should have happened? I don't really know -- that's why I'm asking. I'm sure I have done something wrong in the setup. Just hoping for a pointer in the right direction.

### Code to Reproduce the Issue

```js
// adapter is just an event emitter that accepts a port:
module.exports = class Adapter extends events.EventEmitter {
  constructor ({ port }) {
    super();
    this.port = port;
  }

  doSomething() {
    // call stuff and create an object here
    const obj = getSomeObjectFromSomewhere();
    this.port.write(obj, 'utf8', (err) => {
      if (err) {
        this.emit('error', err);
      }
    });
  }
}

// For now, my mock is really just a constructor calling super
const Adapter = require('../index');

module.exports = class MockAdapter extends Adapter {
  constructor({ port }) {
    super({ port });
  }

  // maybe I setup some stuff for unit testing here for my mock
}

// now here is my unit test
const chai = require('chai');
const MockAdapter = require('./MockAdapter');
const SerialPort = require('serialport/test');

const MockBinding = SerialPort.Binding;
chai.should();

let port = null;
let adapter = null;

describe('adapter', () => {
  beforeEach(() => {
    const portPath = 'COM_ANYTHING'
    MockBinding.createPort(portPath, { echo: true, record: false });
    port = new SerialPort(portPath);

    adapter = new MockAdapter({ port });
  });

  it('should call doSomething() with correct buffer', (done) => {
    const buf = Buffer.alloc(64);
    // do stuff to manually setup buffer here

    // verify that when our adapter writes a value, we get the expected data
    port.on('write', (val) => {
      val.should.be.an('array').and.have.members(buf.toJSON().data);
      done();
    });

    // call doSomething(), which will force adapter to write a buffer to the serial port
    adapter.doSomething();
  });
});

// Code
```

### Versions, Operating System and Hardware

- SerialPort@7.0.2
- Node.js v: 10.6.0
- OS: Mac
- Hardware and chipset: N/A

HipsterBrown, 2018-10-11:
@bdmayes I don't believe there is a `write` event for Serialport ([Reference](https://serialport.io/docs/api-stream#events)). 

The `echo` feature will emit `data` that you write to the SerialPort instance. There is a good example of testing the port writes here -> https://github.com/node-serialport/node-serialport/blob/master/packages/serialport/test/integration.js#L184

Essentially you will need to listen to `port.on('data', callback)` instead of `port.on('write', callback)`.

HipsterBrown, 2018-10-11:
@bdmayes Have you tried listening for an `error` event? Or check out the debugging output -> https://serialport.io/docs/guide-debugging#debugging-output

bdmayes, 2018-10-11:
I wasn't, however I don't seem to be getting any additional output from either one of those changes. I'm trying to set this up with the actual files above in a "sample" directory (ran `rpm init`, then `npm install` some dependencies, etc.). I can see in this project that it's calling the done callback multiple times now because it first fires with a `READY` event. I just fixed that and now it's passing.

```js
it('should call doSomething() with correct buffer', (done) => {
    const buf = Buffer.from('hello');
    // do stuff to manually setup buffer here

    // verify that when our adapter writes a value, we get the expected data
    port.on('data', (data) => {
      if (data.toString('utf8') === 'READY') {
        return;
      }

      data.toString('utf8').should.equal('hello');
      done();
    });

    // call doSomething(), which will force adapter to write a buffer to the serial port
    adapter.doSomething();
  });

I'm not sure why my actual test code wouldn't be emitting debug info, but maybe it's the done callback being called twice there as well? I'll check...
```

HipsterBrown, 2018-10-11:
@bdmayes I'm happy to hear you've figured it out. 

> add in the if statement above to throw away the READY event

You can try using the `ReadyParser` (https://serialport.io/docs/api-parser-ready) to avoid needing the extra if statement.