serialport/node-serialport #1708: readline Hex encoding problem

aminjoharinia, 2018-10-23:
Hi
i have problems reading line with hex encoding
```
1) c0a8
2) fe534d415254434c4f5544aaaa091600010c022d2cd1c0a8
3) fe534d415254434c4f5544aaaa091600010c022d2cd1c0a8
4) fe534d415254434c4f5544aaaa12012b01b8efffffff03fefefe0c2d0223afc0a8
5) fe534d415254434c4f5544aaaa0e012b01b8f036ffff000000d80bc0a8
6) fe534d415254434c4f5544aaaa091600010c022d2cd1c0a8
7) fe534d415254434c4f5544aaaa0653004178e2c0a8
8) fe534d415254434c4f5544aaaa1a150001523128312e3433293c31322d313e5b315d2a002af89ac0a8
9) fe534d415254434c4f5544aaaa1a150002523128312e3433293c31322d323e5b325d2a002ab006c0a8
10) fe534d415254434c4f5544aaaa1a150003523128312e3433293c31322d333e5b335d2a002a8872c0a8
11) fe534d415254434c4f5544aaaa1a150004523128312e3433293c31322d343e5b345d2a002a213ec0a8
12) fe534d415254434c4f5544aaaa1a150005523128312e3433293c31322d353e5b355d2a002a194ac0a8
13) fe534d415254434c4f5544aaaa1a150006523128312e3433293c31322d363e5b365d2a002a51d6c0a8
14) fe534d415254434c4f5544aaaa1a150007523128312e3433293c31322d373e5b375d2a002a69a2c0a8
15) fe534d415254434c4f5544aaaa1a150008523128312e3433293c31322d383e5b385d2a002a136fc0a8
16) fe534d415254434c4f5544aaaa1a150009523128312e3433293c31322d393e5b395d2a002a2b1bc0a8
17) fe534d415254434c4f5544aaaa1a1500
18) 523128312e3433293c31322d31303e5b31305d2ac295c0a8
19) fe534d415254434c4f5544aaaa1a15000b523128312e3433293c31322d31313e5b31315d2a9832c0a8
20) fe534d415254434c4f5544aaaa1a15000c523128312e3433293c31322d31323e5b31325d2a1060c0a8
21) fe534d415254434c4f5544aaaa12012b01b8efffffff03fefefe0c2d0223afc0a8
22) fe534d415254434c4f5544aaaa0e012b01b8f036ffff000000d80bc0a8
23) fe534d415254434c4f5544aaaa091600010c022d2cd1c0a8
24) fe534d415254434c4f5544aaaa091600010c022d2cd1c0a8
25) fe534d415254434c4f5544aaaa12012b01b8efffffff03fefefe0c2d0223afc0a8

```
1) first problem is line 1 and it always happen randomly
2) bigger problem is line 17 & 18 and '0a' is mising ! The complete line is 
   ```
fe534d415254434c4f5544aaaa1a1500`0a`523128312e3433293c31322d31303e5b31305d2ac295c0a8
```
my packets always start wit ```fe534d415254434c4f5544aaaa ```
please help me
Thanks

HipsterBrown, 2018-10-23:
@aminjoharinia 

Can you share some of the relevant code you are using to help us debug the issue?

aminjoharinia, 2018-10-23:
@HipsterBrown 
oh! sorry
```javascript
var SerialPort = require('serialport');
const Readline = require('@serialport/parser-readline')
var port = new SerialPort('/dev/ttyS0', {
baudRate:115200
});

const parser = port.pipe(new Readline({encoding: 'hex'}))
parser.on('data', function(x){
console.log(x)})
```

HipsterBrown, 2018-10-23:
@aminjoharinia I should have mentioned before. Can you also include information about what version of Node and Serialport you are using? And what operating system are you running this code on?

aminjoharinia, 2018-10-23:
@HipsterBrown 
node Version 8.7.0
Serialprt Version 7.0.2
Raspberri pi Raspbian jessie (debian)

HipsterBrown, 2018-10-23:
@aminjoharinia Cool, good to know. 

Have you tried using a different type of delimiter for the Readline parser? 

I might also try splitting after `c0a8` since that is the expected end of each line, it appears. 
```es6
const parser = port.pipe(new Readline({ delimiter: '\r\n' }))
```

As seen in the example here: https://serialport.io/docs/api-parser-readline

aminjoharinia, 2018-10-23:
With This Config  : ```const parser = port.pipe(new Readline({ delimiter: '\r\n' }))``` nothing shows
And with this :  ```const parser = port.pipe(new Readline({ delimiter: '\n' }))``` it shows : 
```
��
�SMARTCLOUD��	
                  -,���
�SMARTCLOUD��	
                  -,���
�SMARTCLOUD��	
                  -,���
�SMARTCLOUD��+��������
                          -#���
�SMARTCLOUD��+��6���
                      ��
�SMARTCLOUD��	
                  -,���
�SMARTCLOUD��	
                  -,���
�SMARTCLOUD��+��������
                          -#���
```
it should be in hex

aminjoharinia, 2018-10-23:
@
```javascript
var SerialPort = require('serialport');
const Readline = require('@serialport/parser-readline')
var port = new SerialPort('/dev/ttyS0', {
baudRate:115200

});

const parser = port.pipe(new Readline({ delimiter: '\n', encoding: 'hex' }))

parser.on('data', function(x){
console.log(x)})
```

aminjoharinia, 2018-10-24:
@HipsterBrown 
Did you find any solution?
I tested other options but i can't read the coming data correctly .... 
and also error happend when use delimiter and encoding together

reconbot, 2018-10-24:
If you log data events of the source data and the parser we can construct a test from the output and determine if the parser is doing something wrong.

```js

var SerialPort = require('serialport');
const Readline = require('@serialport/parser-readline')
var port = new SerialPort('/dev/ttyS0', {
baudRate:115200

});

const parser = port.pipe(new Readline({ delimiter: '\n', encoding: 'hex' }))
port.on('data', data => console.log({ port: data.toString('hex')}))
parser.on('data', data => console.log({ parser: data }))
```

reconbot, 2018-10-24:
Also what hardware are you talking to?

aminjoharinia, 2018-10-26:
@reconbot 
@HipsterBrown 
Hi
Here is what you want 
and my problems solved
```
{ port: 'c0a80afe534d415254434c4f5544aaaa091600010c022d2cd1' }
{ parser: 'fe534d415254434c4f5544aaaa091600010c022d2cd1' }
{ port: 'c0a80afe534d415254434c4f5544aaaa12012b01b8efffffff03fefefe0c2d0223af' }
{ parser: 'fe534d415254434c4f5544aaaa12012b01b8efffffff03fefefe0c2d0223af' }
{ port: 'c0a80afe534d4152' }
{ port: '54434c4f5544aaaa0e012b01b8f036ffff000000d80b' }
{ parser: 'fe534d415254434c4f5544aaaa0e012b01b8f036ffff000000d80b' }
{ port: 'c0a80afe53' }
{ port: '4d415254434c4f5544aaaa091600010c022d2cd1' }
{ parser: 'fe534d415254434c4f5544aaaa091600010c022d2cd1' }
{ port: 'c0a80afe534d' }
{ port: '415254434c4f5544aaaa091600010c022d2cd1' }
{ parser: 'fe534d415254434c4f5544aaaa091600010c022d2cd1' }
{ port: 'c0a80afe53' }
{ port: '4d415254434c4f5544aaaa12012b01b8efffffff' }
{ port: '03fefefe0c2d0223af' }
{ parser: 'fe534d415254434c4f5544aaaa12012b01b8efffffff03fefefe0c2d0223af' }
{ port: 'c0a80afe534d' }
{ port: '415254434c4f5544aaaa0e012b01b8f036ffff000000d80b' }
{ parser: 'fe534d415254434c4f5544aaaa0e012b01b8f036ffff000000d80b' }
{ port: 'c0a80afe534d' }
{ port: '415254434c4f5544aaaa091600010c022d2cd1' }
{ parser: 'fe534d415254434c4f5544aaaa091600010c022d2cd1' }
{ port: 'c0a80afe53' }
{ port: '4d415254434c4f5544aaaa091600010c022d2cd1' }
{ parser: 'fe534d415254434c4f5544aaaa091600010c022d2cd1' }
{ port: 'c0a80afe534d' }
{ port: '415254434c4f5544aaaa091600010c022d2cd1' }
{ parser: 'fe534d415254434c4f5544aaaa091600010c022d2cd1' }
{ port: 'c0a80afe534d' }
{ port: '415254434c4f5544aaaa12012b01b8efffffff03fefefe0c2d0223af' }
{ parser: 'fe534d415254434c4f5544aaaa12012b01b8efffffff03fefefe0c2d0223af' }
{ port: 'c0a80afe534d4152' }
{ port: '54434c4f5544aaaa0e012b01b8f036ffff000000d80b' }
{ parser: 'fe534d415254434c4f5544aaaa0e012b01b8f036ffff000000d80b' }
{ port: 'c0a80afe534d41' }
{ port: '5254434c4f5544aaaa091600010c022d2cd1' }
{ parser: 'fe534d415254434c4f5544aaaa091600010c022d2cd1' }
{ port: 'c0a80afe534d41' }
{ port: '5254434c4f5544aaaa091600010c022d2cd1' }
{ parser: 'fe534d415254434c4f5544aaaa091600010c022d2cd1' }
{ port: 'c0a80afe534d41' }
{ port: '5254434c4f5544aaaa12012b01b8efffffff03fefefe0c2d0223' }
{ port: 'af' }
{ parser: 'fe534d415254434c4f5544aaaa12012b01b8efffffff03fefefe0c2d0223af' }
{ port: 'c0a80afe534d4152' }
{ port: '54434c4f5544aaaa0e012b01b8f036ffff000000d80b' }
{ port: 'c0' }
{ parser: 'fe534d415254434c4f5544aaaa0e012b01b8f036ffff000000d80b' }
{ port: 'a80afe534d41' }
{ port: '5254434c4f5544aaaa091600010c022d2cd1' }
{ parser: 'fe534d415254434c4f5544aaaa091600010c022d2cd1' }
{ port: 'c0a80afe534d41' }
{ port: '5254434c4f5544aaaa091600010c022d2cd1' }
{ parser: 'fe534d415254434c4f5544aaaa091600010c022d2cd1' }
{ port: 'c0a80afe534d' }
{ port: '415254434c4f5544aaaa12012b01b8efffffff03fefefe0c' }
{ port: '2d0223af' }
{ parser: 'fe534d415254434c4f5544aaaa12012b01b8efffffff03fefefe0c2d0223af' }
{ port: 'c0a80afe534d41' }
{ port: '5254434c4f5544aaaa0e012b01b8f036ffff000000d80b' }
{ parser: 'fe534d415254434c4f5544aaaa0e012b01b8f036ffff000000d80b' }
{ port: 'c0a80afe534d41' }
{ port: '5254434c4f5544aaaa091600010c022d2cd1' }
{ parser: 'fe534d415254434c4f5544aaaa091600010c022d2cd1' }
{ port: 'c0a80afe534d4152' }
{ port: '54434c4f5544aaaa091600010c022d2cd1' }
{ parser: 'fe534d415254434c4f5544aaaa091600010c022d2cd1' }
{ port: 'c0a80afe534d41' }
{ port: '5254434c4f5544aaaa12012b01b8efffffff03fefefe0c2d' }
{ port: '0223af' }
{ parser: 'fe534d415254434c4f5544aaaa12012b01b8efffffff03fefefe0c2d0223af' }
{ port: 'c0a80afe534d41' }
{ port: '5254434c4f5544aaaa0e012b01b8f036ffff000000d80b' }
{ parser: 'fe534d415254434c4f5544aaaa0e012b01b8f036ffff000000d80b' }
{ port: 'c0a80afe534d41' }
{ port: '5254434c4f5544aaaa0653004178e2' }
{ parser: 'fe534d415254434c4f5544aaaa0653004178e2' }
{ port: 'c0a80afe534d41' }
```
 1 ) I found '0a' "The line feed" was the responsible for parser going to next line , i changed delimiter to 'c0a80a' and now it perfectly works 

2 )And for the error 
```
TypeError: "delimiter" has a 0 or undefined length
```
when the encoding is hex , i should not use  '\n' , '\r' , if we use 'hex' it works correctly

>Another Question : how could i write 'hex' ?
Is it correct?
The system does not show anything ( system is SBus Home Automation System)
```
port.write(Buffer.from('aaaa030130'));
```

Thanks a lot for your helps and answers 👍

reconbot, 2018-10-26:
```js
> Buffer.from('aaaa030130')
<Buffer 61 61 61 61 30 33 30 31 33 30>
```
☝️ uses utf8 by default 

To use HEX [pass the encoding on the second argument](https://nodejs.org/api/buffer.html#buffer_class_method_buffer_from_string_encoding).
```
> Buffer.from('aaaa030130', 'hex')
<Buffer aa aa 03 01 30>
```

Glad we could help!

closed 2018-10-26 by reconbot