serialport/node-serialport #1709: SyntaxError when trying to use node-serialport on RPI 2

gbrennon, 2018-10-24:
### Summary of Problem
Hey guys,

I'm trying to use node-serialport on a RPI 2 with NVM using Node v8.12.0.

Well, the package installation is done and it didn't returns me any error, but when I try to run my node script it returns me an "SyntaxError" coming from "@serialport/bindings/lib/linux.js:16".

This reference in the file points to a class definition(class LinuxBinding extends AbstractBinding) and it is telling me that "Block-scoped declarations (let, const, function, class) not yet supported outside strict mode".

How can I fix this?

### Code to Reproduce the Issue

```js
var serialport = new SerialPort(path, {
  baudRate: 9600,
});

port.on('open', function() {
  port.write('main screen turn on', function(err) {
    if (err) {
      return console.log('O cabo usb não está conectado: ', err.message);
    }
    console.log('message written');
  });
});

// Code
```

### Versions, Operating System and Hardware
- SerialPort: ^7.0.2
- Node.js: v8.12.0
- Ubuntu Mate Linux

HipsterBrown, 2018-10-24:
@gbrennon Are you certain you have Node v8.12.0 running on your Raspberry Pi? Otherwise you can try adding `'use strict';` to the top of your main file that is requiring SerialPort.

Does this code run without errors on your device?
```es6
const test = 'test';
console.log(test);
```

aminjoharinia, 2018-10-26:
Hi
First on Raspberry you should enable Serial
Second you should choose your path 
on RPI 3 is ```ttyS0``` on RPI zero ```ttyAMA0```
example :
 ```
var port = new SerialPort('/dev/ttyS0', {
baudRate:115200
});
```

gbrennon, 2018-10-30:
@HipsterBrown I'm calling the program using "sudo" so maybe it isn't calling my NVM selected version... I'll look which version of node is being called when used with sudo

gbrennon, 2018-10-31:
@HipsterBrown thank u man! It is working now! The problem was because the node version was wrong for the sudo user.

closed 2018-10-31 by gbrennon