serialport/node-serialport #1711: error if i try to run

praveen5655, 2018-10-27:
## 👉 [Please follow one of these issue templates](https://github.com/node-serialport/node-serialport/issues/new/choose) 👈
Uncaught Error: A dynamic link library (DLL) initialization routine failed.
\\?
\C:\Users\praveen\Desktop\ELECTRON\dental_chair_start\node_modules\@serialport\bindings\build\Release\bindings.node

reconbot, 2018-10-27:
Hi
Please check out the electron install guide
https://serialport.io/docs/guide-installation#electron

And then fill a new issue using this form with more details
https://github.com/node-serialport/node-serialport/issues/new?template=support.md

Thanks!

closed 2018-10-27 by reconbot

hanfengcan, 2018-10-28:
https://github.com/FakeFullStack/QA/issues/2 

here, I write chinese doc to rebuild serialport in electron.

reconbot, 2018-10-28:
Thank you @hanfengcan! If I add Chinese to https://serialport.io can you help translate?

hanfengcan, 2018-10-29:
OK, I'll find a time to translate

reconbot, 2018-11-03:
@hanfengcan I have made a crowdin account to help https://crowdin.com/translate/node-serialport/78/en-zhcn

I am working on the build tools on #1718