serialport/node-serialport #1713: How to use with nw.js and webpack

Arti3DPlayer, 2018-10-29:
I read all articles https://github.com/node-serialport/node-serialport/search?q=webpack&type=Issues&utf8=%E2%9C%93

but din't find answer. When I use serialport as external, how do I can include it to build ?

In nw.js my entrypoint is `index.html` inside it I include `bundle.js` generated by with `webpack`.

HipsterBrown, 2018-10-29:
@Arti3DPlayer Have you tried the answer shown here -> https://github.com/railsware/bozon/issues/40#issuecomment-287992554

Also, have you read the [build instructions for serialport to target nw.js](https://serialport.io/docs/guide-installation#nwjs)?

Arti3DPlayer, 2018-10-29:
@HipsterBrown Yes. I added file `.prebuildrc`
```
build_from_source=true
runtime=node-webkit
target=0.28.3
```

Then do `npm install --save nodeserial`. But what now. How to use it in project ?

I run nwjs like: `/path/to/nw /path/to/package.json` where i have index.html file.
And
`import * as SerialPort from 'serialport';`

Is not working for me, because I didn't include it in webpack

HipsterBrown, 2018-10-30:
@Arti3DPlayer I need to see more code to be able to help. Do you have a repo you could point to?

Arti3DPlayer, 2018-10-30:
@HipsterBrown oh, yes, I have the same structure as you shown. I will try. Thank you for your help. Maybe you need to attach this in docs:)

Arti3DPlayer, 2018-10-30:
But wait if, webpack will ignore bundling it, when I did packaging of my app without node_modules folder, how do I can use it then. The main reason why I use webpack - I want to bundle all js to one file and then do protection of this file https://github.com/nwjs/nw.js/wiki/protect-javascript-source-code-with-v8-snapshot

reconbot, 2018-10-30:
Unclear if that's compatible with binary modules. Serialport's JS can be
bundled but it's binary module cannot because it's not JS.

Arti3DPlayer, 2018-10-30:
ok, thank you. I will provide details later, when do more research

closed 2018-10-30 by Arti3DPlayer