serialport/node-serialport #1714: Serial is breaking lines in a wrong way...

gbrennon, 2018-10-31:
### Summary of Problem
So,
I'm sending data through the serial to my RPi using this package and it isn't working properly when reading the serial port...

When reading the serial through MINICOM, SERIAL MONITOR or other tools like this it reads the serial the way it should be reading... But  when I'm reading the messages via the module it is just messed up.

I'm sending my messages from an Arduino that receive data from device sensors like this using the Serial.println:
- "Device: value"
- "Device: value"
- "Device: value"

But when reading the serial in the RPi the messages are coming randomly broken... Like it was finding some "\n" in the mid of the Serial prints, telling the bus that the write is complete.
- "Device: valueDevice:value"
- "Device:"
- "val"
- "ue"
- "Device: valueDe"

Any clues of what this could be or how to solve this?

### Code to Reproduce the Issue

```js
var port = new SerialPort('/dev/ttyUSB0', {
  baudRate: 9600,
});
port.on('data', function(data) {
  //it should print each sent message
  console.log(data);
});
// Code
```

### Versions, Operating System and Hardware

- SerialPort@7.0.2
- Node.js v8.2.0
- Linux
- Hardware and chipset? Prolific

reconbot, 2018-10-31:
Give this a try

https://serialport.io/docs/api-parser-readline

fsoarezperu, 2018-11-05:
im having the same issue, was it sovled? any suggestions?

reconbot, 2018-11-05:
Same suggestion. Data events don't know anything about what's in them, they are emitted as they are received. You need a parser if you'd like to split on newlines or other bytes.

closed 2019-01-11 by stale[bot]