serialport/node-serialport #1715: [windows] Serialport.write() will be blocked when writing []

fp901105, 2018-11-02:
### Summary of Problem

(Please answer all 3)

- What are you trying to do?
I am trying to upgrade the version from 4.0.7 to 7.0.2
I am using Window 7, but the same code works in Ubuntu 16.04
- What happens?
Serialport.write() function get stuck when write a null array to serialport
- What should have happened?
Multiple writes should be successfully executed
### Code to Reproduce the Issue

```js
var serialport_driver = require('serialport');

class test_port{
  port_name;
  port;
  open;
  write;
  drain;
} 

test_port.prototype.open = ()=>{
  this.port = new serialport_driver('COM1',{baudRate:115200, autoOpen:false})
  this.port.open(error=>{
    console.log('port opened')
  })
}

test_port.prototype.write = (data)=>{
  this.port.write(data, (function (_this) {
    return function (error: any, results: any) {
      console.log('written data---',data)
      if (error) {
        return _this._failure(error);
      }
    };
  })(this));
}

test_port.prototype.drain = ()=>{
  this.port.drain(error=>{
    console.log('drained')
  })
}

var arr = Buffer.from([])
var buf = Buffer.from('this is a test');
var buf2 = Buffer.from('this is a test');

var testport = new test_port();
testport.open()
testport.write(arr)
testport.write(buf)
testport.write(buf2)

// Code
The expected output should be:
opened
written----:  <Buffer >
written----:  <Buffer 74 68 69 73 20 69 73 20 61 20 74 65 73 74>
written----:  <Buffer 74 68 69 73 20 69 73 20 61 20 74 65 73 74>

but get stuck at 
opened

```

### Versions, Operating System and Hardware

- SerialPort@7.0.2
- Node.js v8.9.4
- Windows?
- Hardware and chipset? Other

HipsterBrown, 2018-11-02:
@fp901105 Can you share the debug output from your code? https://serialport.io/docs/guide-debugging#cmd

reconbot, 2018-11-02:
This isn't a defined case in our code afaict, we should probably do nothing (matches the other io streams from node core).

```
> process.stdout.write(Buffer.from([]))
true
```

HipsterBrown, 2018-11-02:
So after digging into the windows bindings code a bit, I found the following interesting bit of info that could be relevant to this issue. 

[Windows WriteFileEx function](https://docs.microsoft.com/en-us/windows/desktop/api/fileapi/nf-fileapi-writefileex)
> `nNumberOfBytesToWrite`
> A value of zero specifies a null write operation. The behavior of a null write operation depends on the underlying file system.

It could also be related to the logic in `WriteIOCompletion` (https://github.com/node-serialport/node-serialport/blob/master/packages/bindings/src/serialport_win.cpp#L331-L346). I don't think the write will ever complete if the bufferLength is 0. So we would need to check if `bytesWritten` equals `baton->bufferLength` to mark the write complete. 

/cc: @reconbot

HipsterBrown, 2018-11-02:
@reconbot I think it's related to the `WriteIOCompletion` routine in the windows bindings (https://github.com/node-serialport/node-serialport/blob/master/packages/bindings/src/serialport_win.cpp#L331-L346)

Also unsure of how `WriteFileEx` handles null writes (https://docs.microsoft.com/en-us/windows/desktop/api/fileapi/nf-fileapi-writefileex)

> `nNumberOfBytesToWrite`
> A value of zero specifies a null write operation. The behavior of a null write operation depends on the underlying file system.

reconbot, 2018-11-02:
we should just special case an empty buffer

PR #1716 (2018-11-05): Early write callback for empty buffers
Fixes #1715 

As discussed with @reconbot, the native bindings for write operations have different behavior for empty buffers. So it's probably better to just callback early and not bother the bindings layer with the empty buffer.
closed 2018-11-05 by HipsterBrown