serialport/node-serialport #1721: write to UART on raspberry apparently enters an infinite loop

juanjo75es, 2018-11-09:
### Summary of Problem

I am trying to communicate through UART with an E32 TTL 100 LORA module from a Raspberri Pi 3 B+. 
When I write to the port there is no error but the call doesn't return. Apparently it enters an infinite loop and the blackberry becomes irresponsive. It should return or produce an exception with an error.

### Code to Reproduce the Issue

```js
var pins=require("pi-pins")

var pin1 = pins.connect(17),
    pin2 = pins.connect(27);

pin1.mode('out');
pin2.mode('out'); 
pin1.value(1);
console.log("pin1:", pin1.value());
pin2.value(1);
console.log("pin2:", pin2.value());

const SerialPort = require('serialport')
const port = new SerialPort('/dev/ttyAMA0', {
  baudRate: 9600,  
  dataBits: 8,
  stopBits: 1,
  parity: 'none',
  autoOpen: false
},function(err){
  console.log(err);
})

setTimeout(function(){
  port.open(function (err) {
    if (err) {
      return console.log('Error opening port: ', err.message)
    }

  })
},5000);

// The open event is always emitted
port.on('open', function() {
  // open logic
  console.log('opened ')
  var buf = new Buffer([ 0xC1, 0xC1, 0xC1 ]);
  port.write(buf, function(err,n) {
    if (err) {
      return console.log('Error on write: ', err.message)
    }
    console.log('message written')
  })
  
})

// Code
```
The console writes 'opened' and nothing more.

### Versions, Operating System and Hardware

- SerialPort@7.0.2
- Node.js v9.11.1
- Raspbian 9 (stretch)
- Hardware and chipset Raspberry Pi 3 B+

HipsterBrown, 2018-11-09:
@juanjo75es Can you try listening for an `error` event on the `port`, i.e. ```port.on('error', console.error)```?

Also, can you share the debug logs? https://serialport.io/docs/en/guide-debugging#linux-osx

juanjo75es, 2018-11-09:
> if you search the issues a bit, you'll find it's usually because the rpi's serial hardware isn't setup correctly
> […](#)

yep, I already guessed it's caused by the hardware setup but I thought you'd like to know that the write function doesn't return and doesn't give any error. I guess that's an issue. Didn't find any issue with a similar topic. It would be helpful if it reported the problem so that I could guess what the problem is with the setup.

I tried listening the error event. Didn't get any error either. The function doesn't return.  Maybe I'll try later debugging ... 
Someone told me that:

> The way we handle this is to send one command at a time and wait for a response before sending the next command. We send a command and set a timeout that the device must respond within. If we receive a response that makes some kind of sense, then we move on. If the response doesn't make sense or we don't receive a response in a specified amount of time, then we throw an exception. So commands are sent in a serial manner, one at a time and we ensure we get a response. We deal with up to 28 ports simultaneously and keep track of them in an array. We also use sockets to communicate with a browser window, so we can have up to 28 browser windows open simultaneously, talking to 28 ports. We tack the socket ID on to the serial port info so when we get a response, we know which browser window it came from and then gets sent back to. The limitation of 28 ports is not a real limitation but the number of ports we could physically connect to one computer before we ran out of hardware. 

Apparently that timer is not working.
Will continue invstigating ...

Thanks

juanjo75es, 2018-11-10:
Update:
If I disconnect the module the write function still doesn't return. 
My problem apparently was that the device for UART in Raspberry 3 B+ is /dev/serial0 and /dev/ttyAMA0 is now used for bluetooth.

HipsterBrown, 2018-11-10:
@juanjo75es Does the write function work when you use the `/dev/serial0` port now?

HipsterBrown, 2018-11-12:
@juanjo75es Do you have an example of the code you're using to read back the response?

juanjo75es, 2018-11-12:
```
serialPort.on('data', function(data){
        console.log(data);
        readData += data.toString();
        
    });
```
Apparently there was an incompatibility issue with library pi-pins. It's working now.

closed 2018-11-12 by juanjo75es