serialport/node-serialport #1722: Build Bindings fails from docker node:10 on armv7

Jackbennett, 2018-11-11:
### Summary of Problem

(Please answer all 3)

- What - Trying to build @serialport/bindings
- What happens? - Lots of warning/error output but something finishes.
- What should have happened? No errors and serialport should work.

Before I fill up this issue with build logs can I idiot-check the following?

### Versions, Operating System and Hardware

```json
    "@serialport/bindings": "^2.0.2",
    "@serialport/parser-readline": "^2.0.1",
    "@serialport/stream": "^2.0.1",
```
- `Docker pull node:10`
- RPI3 Hypriot 1.9 armv7, uart pins

Are we supposed to be able to build this module from the provided docker hub node image, I can't find it documented yes or no but the impression I get is the node images should be good to go compiling native modules.

If I run the container this makes everything executes successfully but I never see any data events after open. Now that could be some wiring on my end but I'm trying to rule out if serialport can  run with a binding that's not working. The docker run is; `docker run --rm -it --privileged myImage node ./test/readport.js`

### build log output
```
Sending build context to Docker daemon  217.1kB
Step 1/10 : FROM node:10 as build
 ---> e69cb89724b8
Step 2/10 : WORKDIR /app
 ---> Using cache
 ---> 81c7f7f72086
Step 3/10 : COPY ./package*.json ./
 ---> efefb06bfcd6
Step 4/10 : RUN npm install
 ---> Running in 6a163d47ea9c

> epoll@2.0.6 install /app/node_modules/epoll
> node-gyp rebuild

make: Entering directory '/app/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
  SOLINK_MODULE(target) Release/obj.target/epoll.node
  COPY Release/epoll.node
make: Leaving directory '/app/node_modules/epoll/build'

> @serialport/bindings@2.0.2 install /app/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=10.13.0 runtime=node arch=arm platform=linux)
make: Entering directory '/app/node_modules/@serialport/bindings/build'
  CXX(target) Release/obj.target/bindings/src/serialport.o
In file included from ../../../nan/nan_new.h:189:0,
                 from ../../../nan/nan.h:222,
                 from ../src/./serialport.h:6,
                 from ../src/serialport.cpp:1:
../../../nan/nan_implementation_12_inl.h: In static member function 'static Nan::imp::FactoryBase<v8::StringObject>::return_t Nan::imp::Factory<v8::StringObject>::New(v8::Local<v8::String>)':
../../../nan/nan_implementation_12_inl.h:340:37: warning: 'static v8::Local<v8::Value> v8::StringObject::New(v8::Local<v8::String>)' is deprecated (declared at /root/.node-gyp/10.13.0/include/node/v8.h:5052): Use Isolate* version [-Wdeprecated-declarations]
   return v8::StringObject::New(value).As<v8::StringObject>();
                                     ^
In file included from ../src/./serialport.h:6:0,
                 from ../src/serialport.cpp:1:
../../../nan/nan.h: In constructor 'Nan::Utf8String::Utf8String(v8::Local<v8::Value>)':
../../../nan/nan.h:1066:53: warning: 'v8::Local<v8::String> v8::Value::ToString() const' is deprecated (declared at /root/.node-gyp/10.13.0/include/node/v8.h:10040): Use maybe version [-Wdeprecated-declarations]
       v8::Local<v8::String> string = from->ToString();
                                                     ^
../../../nan/nan.h:1080:74: warning: 'int v8::String::WriteUtf8(char*, int, int*, int) const' is deprecated (declared at /root/.node-gyp/10.13.0/include/node/v8.h:2658): Use Isolate* version [-Wdeprecated-declarations]
         length_ = string->WriteUtf8(str_, static_cast<int>(len), 0, flags);
                                                                          ^
../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE Open(Nan::NAN_METHOD_ARGS_TYPE)':
[... 78 lines cut ...]
       v8::Local<v8::String> string = from->ToString();
                                                     ^
../../../nan/nan.h:1080:74: warning: 'int v8::String::WriteUtf8(char*, int, int*, int) const' is deprecated (declared at /root/.node-gyp/10.13.0/include/node/v8.h:2658): Use Isolate* version [-Wdeprecated-declarations]
         length_ = string->WriteUtf8(str_, static_cast<int>(len), 0, flags);
                                                                          ^
../src/poller.cpp: In static member function 'static void Poller::onData(uv_poll_t*, int, int)':
../src/poller.cpp:69:29: warning: 'v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const' is deprecated (declared at ../../../nan/nan.h:1655) [-Wdeprecated-declarations]
   obj->callback.Call(2, argv);
                             ^
  CXX(target) Release/obj.target/bindings/src/serialport_linux.o
  SOLINK_MODULE(target) Release/obj.target/bindings.node
  COPY Release/bindings.node
make: Leaving directory '/app/node_modules/@serialport/bindings/build'
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN hardware@1.0.0 No repository field.

added 81 packages from 54 contributors and audited 145 packages in 62.093s
found 0 vulnerabilities

Removing intermediate container 6a163d47ea9c
 ---> daaf1d9603bd
Step 5/10 : FROM node:10-slim

... etc
```

HipsterBrown, 2018-11-12:
@Jackbennett Can you share the code you're using in your `readport.js` script? 

I'm not sure how Docker virtualizes the serial port access. Are you using the [official `node:10` Docker image](https://hub.docker.com/_/node/) instead of the [`arm32v7/node:10` image](https://hub.docker.com/r/arm32v7/node/)? It appears the latter image is the one recommended by the Hypriot project.

Jackbennett, 2018-11-12:
Looking at the imageID's docker has pulled the correct node arch when I built it on the PI

```
arm32v7/node                                    10-slim             8499ba3f9847        10 days ago         157MB
node                                            10-slim             8499ba3f9847        10 days ago         157MB
arm32v7/node                                    10                  e69cb89724b8        10 days ago         555MB
node                                            10                  e69cb89724b8        10 days ago         555MB
```

The readport file looks like;
```javascript
const SerialPort = require('@serialport/stream')
const Readline = require('@serialport/parser-readline')
const Binding = require('@serialport/bindings')

SerialPort.Binding = Binding

const port = new SerialPort('/dev/ttyAMA0')
const parser = port.pipe(new Readline({ delimiter: "\r\n", encoding: 'ascii' }))

port.on('open', () => {
  console.log(`Listening for data events on serial port`)
})
port.on('data', raw => {
  console.log(`raw data event log: ${raw}`)
})

parser.on('error', e => console.error(e))
parser.on('data', d => {
  console.log(`${Date()} Seriport logged: ${d}`)
})
parser.on('open', () => {
  console.log(`Parser is ready to process data events on \r\n`)
})
```

the only output produced is "Listening for data events on serial port"

seems like just the same build output
```
Step 4/10 : RUN npm install
 ---> Running in 62d94e1bd77e

> epoll@2.0.6 install /app/node_modules/epoll
> node-gyp rebuild

make: Entering directory '/app/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
  SOLINK_MODULE(target) Release/obj.target/epoll.node
  COPY Release/epoll.node
make: Leaving directory '/app/node_modules/epoll/build'

> @serialport/bindings@2.0.2 install /app/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=10.13.0 runtime=node arch=arm platform=linux)
make: Entering directory '/app/node_modules/@serialport/bindings/build'
  CXX(target) Release/obj.target/bindings/src/serialport.o
In file included from ../../../nan/nan_new.h:189:0,
                 from ../../../nan/nan.h:222,
                 from ../src/./serialport.h:6,
                 from ../src/serialport.cpp:1:
../../../nan/nan_implementation_12_inl.h: In static member function 'static Nan::imp::FactoryBase<v8::StringObject>::return_t Nan::imp::Factory<v8::StringObject>::New(v8::Local<v8::String>)':
../../../nan/nan_implementation_12_inl.h:340:37: warning: 'static v8::Local<v8::Value> v8::StringObject::New(v8::Local<v8::String>)' is deprecated (declared at /root/.node-gyp/10.13.0/include/node/v8.h:5052): Use Isolate* version [-Wdeprecated-declarations]
   return v8::StringObject::New(value).As<v8::StringObject>();
                                     ^
In file included from ../src/./serialport.h:6:0,
                 from ../src/serialport.cpp:1:
../../../nan/nan.h: In constructor 'Nan::Utf8String::Utf8String(v8::Local<v8::Value>)':
../../../nan/nan.h:1066:53: warning: 'v8::Local<v8::String> v8::Value::ToString() const' is deprecated (declared at /root/.node-gyp/10.13.0/include/node/v8.h:10040): Use maybe version [-Wdeprecated-declarations]
[... 78 lines cut ...]
../../../nan/nan_implementation_12_inl.h:340:37: warning: 'static v8::Local<v8::Value> v8::StringObject::New(v8::Local<v8::String>)' is deprecated (declared at /root/.node-gyp/10.13.0/include/node/v8.h:5052): Use Isolate* version [-Wdeprecated-declarations]
   return v8::StringObject::New(value).As<v8::StringObject>();
                                     ^
In file included from ../src/poller.cpp:1:0:
../../../nan/nan.h: In constructor 'Nan::Utf8String::Utf8String(v8::Local<v8::Value>)':
../../../nan/nan.h:1066:53: warning: 'v8::Local<v8::String> v8::Value::ToString() const' is deprecated (declared at /root/.node-gyp/10.13.0/include/node/v8.h:10040): Use maybe version [-Wdeprecated-declarations]
       v8::Local<v8::String> string = from->ToString();
                                                     ^
../../../nan/nan.h:1080:74: warning: 'int v8::String::WriteUtf8(char*, int, int*, int) const' is deprecated (declared at /root/.node-gyp/10.13.0/include/node/v8.h:2658): Use Isolate* version [-Wdeprecated-declarations]
         length_ = string->WriteUtf8(str_, static_cast<int>(len), 0, flags);
                                                                          ^
../src/poller.cpp: In static member function 'static void Poller::onData(uv_poll_t*, int, int)':
../src/poller.cpp:69:29: warning: 'v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const' is deprecated (declared at ../../../nan/nan.h:1655) [-Wdeprecated-declarations]
   obj->callback.Call(2, argv);
                             ^
  CXX(target) Release/obj.target/bindings/src/serialport_linux.o
  SOLINK_MODULE(target) Release/obj.target/bindings.node
  COPY Release/bindings.node
make: Leaving directory '/app/node_modules/@serialport/bindings/build'
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN hardware@1.0.0 No repository field.

added 81 packages from 54 contributors and audited 145 packages in 81.564s
found 0 vulnerabilities
```

HipsterBrown, 2018-11-12:
What is your Pi connected to that you're expecting data from? 

If the port is not throwing an errors and the `open` event is emitted, that leads me to believe the build for the bindings succeeded.

Jackbennett, 2018-11-13:
So the build warnings seem to be just that, warning, you're right it's opening the port without issue.

I hadn't reconfigured the PI3 to remove the linux console and BlueTooth from the uart to enable pins 14 and 15.

Thanks for ruling out serialport.

closed 2018-11-13 by Jackbennett