serialport/node-serialport #1726: Open always returns "Error: Port is opening" under Raspberry Pi 3

pguillem, 2018-11-21:
Whenever I try to access any serial device, the "open" action always returns an error saying "Port is Opening" and just sits there. This happens no matter what serial device I try to access on the Raspberry... in this case, a Freescale sensor properly enumerated.

* SerialPort.list() correctly returns 3 devices in the list.
* I can access all devices perfectly from MINICOM or just doing a CAT on the tty device.

Minibian 4.1.18 (Debian)
Node version 10.13.0
NPM version: 6.4.1
SerialPort version: 7.0.2

Some code

`
var SerialPort = require("serialport");
var portName = "/dev/ttyACM0"; 

var sp = new SerialPort(
portName, {
        parser: SerialPort.parsers.readline,
        baudRate: 230400
});

sp.open(function (error) {
  if ( error ) {
    console.log(''+error);
  } else {
    console.log('Freescale abierto :)');
    sp.on('data', sendSerialData); 
 }
});
`

#node freescale.js

Output:

Error: Port is Opening

pguillem, 2018-11-21:
Ok..

I commented the "if(port.opening)" conditional at line **stream.js** but started getting the "cannot lock port" problem... 

then I followed your suggestion on #1426 and used "lock:false" in the constructor... and THEN the data started coming in. 

Weird... there is something definetly not working as it should. The this.opening property on the raspberry perhaps...

reconbot, 2018-11-21:
What version are you using? Looks really old

pguillem, 2018-11-21:
Minibian 4.1.18 (Debian)
Node version 10.13.0
NPM version: 6.4.1
SerialPort version: 7.0.2

pguillem, 2018-11-21:
I have tried going 9600 baud and a set of data every 0.5 segs to make sure is not some UART error... but it just adds random repetitions everywhere and eats some parts of the data.... any ideas?

pguillem, 2018-11-22:
Actually... that looks very much as a simple duplicate instance of every letter of the original string.,, with the exception that some duplicate chunks are actually whole syllables.

reconbot, 2018-11-22:
Raspberry pi has some conflating of the serial and the console. Are you
sure the js data is doubled and maybe not just your output however you're
consoled into the pie?

reconbot, 2018-11-22:
Also are you perhaps logging twice? Seeing the code can help.

pguillem, 2018-11-22:
I´m also broadcasting the serial messages to a websocket... the consumer in the web part sees the same doubled string. 100% sure.

Sure thing, here is the code

```
var SerialPort = require("serialport");
const Readline = require('@serialport/parser-readline')
var WebSocketServer = require('ws').Server;

var SERVER_PORT = 8080;                                 // Puerto para el servidor
var wss = new WebSocketServer({port: SERVER_PORT});     // Servidor Websocket
var connections = new Array;                            // listado de conexiones entrantes
var portName = "/dev/serial/by-id/usb-P_E_Microcomputer_Systems_Inc._OpenSDA_Hardware_SDA4F5F1E7E-if00"; // Puerto serial del sensor freescale (Siempre debe ser enumer$

//Creamos conn sin lock puerto Serial
var sp = new SerialPort(
        portName, {
                baudRate: 9600,
                lock: false
        }, true);

const parser = sp.pipe(new Readline({ delimiter: '\r\n' })); //Definimos a partir de donde es una nueva linea para el parser
sp.pipe(parser); // Conectamos los eventos de SP al Parser

sp.open(function (error) {
  if ( error ) {
    //console.log(''+error);
  } else {
 //console.log('Freescale abierto :)');
    //sp.on('data', sendSerialData);
    parser.on('data', sendSerialData); //Si hay datos.. haga algo con ellos, pero procesados por el parser
  }
});

function sendSerialData(data) {
  console.log(String(data));
  if (connections.length > 0) {
    broadcast(String(data));    //Distribuimos mensaje a todos los webs conectados
  }
}

```

reconbot, 2018-11-22:
you're piping to the parser twice

pguillem, 2018-11-22:
Jesus christ..... you are absolutely right.
Fixed....

Thank you SO much man!

reconbot, 2018-11-22:
happy to help =)

pguillem, 2018-11-22:
You sir. simply rock!

closed 2018-11-22 by pguillem